gwlim/openwrt-sfe-flowoffload-ath79 #2: Wireless connection unstable on Archer C7v2

demorphica, 2018-12-05:
Hi, thank you very much for doing this work
I am using this on Archer C7v2 & was using your older image that got deprecated. 
I faced issues with wireless stability on the older image, both 5Ghz and 2Ghz networks would become disabled after some time
After upgrading to the new image i still see the same problem, i think it got worse since 5GHz stays disabled and 2GHz drops connections after just a few seconds 

Kernel log

```
[    0.000000] Linux version 4.14.82 (gwlim@openwrt.org) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r8635-e261c8b764)) #0 Tue Dec 4 22:39:53 2018
[    0.000000] MyLoader: sysp=41f93c02, boardp=834d3095, parts=6eef582c
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 804826d0, node_mem_map 81000020
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] random: get_random_bytes called from 0x80486780 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line:  board=ARCHER-C7-V2  console=ttyS0,115200 rootfstype=squashfs noinitrd
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 124772K/131072K available (3602K kernel code, 173K rwdata, 464K rodata, 296K init, 196K bss, 6300K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] Clocks: CPU:720.000MHz, DDR:600.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000007] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008284] Calibrating delay loop... 359.42 BogoMIPS (lpj=718848)
[    0.042883] pid_max: default: 32768 minimum: 301
[    0.047924] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.054970] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.064160] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.074560] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.081419] NET: Registered protocol family 16
[    0.087231] MIPS: machine is TP-LINK Archer C7
[    0.095074] ar724x-pci ar724x-pci.0: PCIe link is down
[    0.100568] registering PCI controller with io_map_base unset
[    0.106815] registering PCI controller with io_map_base unset
[    0.545661] Can't analyze schedule() prologue at 803df9d8
[    0.560505] PCI host bridge to bus 0000:00
[    0.564911] pci_bus 0000:00: root bus resource [mem 0x10000000-0x11ffffff]
[    0.572227] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.578155] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.585379] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.593826] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.593935] PCI host bridge to bus 0000:01
[    0.598313] pci_bus 0000:01: root bus resource [mem 0x12000000-0x13ffffff]
[    0.605632] pci_bus 0000:01: root bus resource [io  0x0001]
[    0.611566] pci_bus 0000:01: root bus resource [??? 0x00000000 flags 0x0]
[    0.618794] pci_bus 0000:01: No busn resource found for root bus, will use [bus 01-ff]
[    0.627251] pci 0000:01:00.0: [168c:003c] type 00 class 0x028000
[    0.627307] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x001fffff 64bit]
[... 1020 lines cut ...]
Wed Dec  5 21:41:11 2018 kern.info kernel: [ 2567.933134] br-lan: port 2(wlan1) entered blocking state
Wed Dec  5 21:41:11 2018 kern.info kernel: [ 2567.938560] br-lan: port 2(wlan1) entered forwarding state
Wed Dec  5 21:41:13 2018 daemon.info hostapd: wlan1: STA b0:7f:b9:44:70:1c IEEE 802.11: authenticated
Wed Dec  5 21:41:13 2018 daemon.info hostapd: wlan1: STA b0:7f:b9:44:70:1c IEEE 802.11: associated (aid 1)
Wed Dec  5 21:41:15 2018 daemon.info hostapd: wlan1: STA b0:7f:b9:44:70:1c RADIUS: starting accounting session 41894436A3420BD0
Wed Dec  5 21:41:15 2018 daemon.info hostapd: wlan1: STA b0:7f:b9:44:70:1c WPA: pairwise key handshake completed (RSN)
Wed Dec  5 21:41:16 2018 daemon.info odhcpd[1515]: Using a RA lifetime of 0 seconds on br-lan
Wed Dec  5 21:41:17 2018 daemon.info dnsmasq-dhcp[2320]: DHCPDISCOVER(br-lan) b0:7f:b9:44:70:1c
Wed Dec  5 21:41:17 2018 daemon.info dnsmasq-dhcp[2320]: DHCPOFFER(br-lan) 10.1.1.254 b0:7f:b9:44:70:1c
Wed Dec  5 21:41:17 2018 daemon.info dnsmasq-dhcp[2320]: DHCPREQUEST(br-lan) 10.1.1.254 b0:7f:b9:44:70:1c
Wed Dec  5 21:41:17 2018 daemon.info dnsmasq-dhcp[2320]: DHCPACK(br-lan) 10.1.1.254 b0:7f:b9:44:70:1c d3m0rphlcaext
Wed Dec  5 21:41:19 2018 daemon.info hostapd: wlan1: STA 20:a6:0c:3c:09:2d IEEE 802.11: authenticated
Wed Dec  5 21:41:19 2018 daemon.info hostapd: wlan1: STA 20:a6:0c:3c:09:2d IEEE 802.11: associated (aid 2)
Wed Dec  5 21:41:20 2018 daemon.info hostapd: wlan1: STA 20:a6:0c:3c:09:2d RADIUS: starting accounting session FAF4555C3E49BCB3
Wed Dec  5 21:41:20 2018 daemon.info hostapd: wlan1: STA 20:a6:0c:3c:09:2d WPA: pairwise key handshake completed (RSN)
Wed Dec  5 21:41:20 2018 daemon.info dnsmasq-dhcp[2320]: DHCPDISCOVER(br-lan) 20:a6:0c:3c:09:2d
Wed Dec  5 21:41:20 2018 daemon.info dnsmasq-dhcp[2320]: DHCPOFFER(br-lan) 10.1.1.11 20:a6:0c:3c:09:2d
Wed Dec  5 21:41:20 2018 daemon.info dnsmasq-dhcp[2320]: DHCPREQUEST(br-lan) 10.1.1.11 20:a6:0c:3c:09:2d
Wed Dec  5 21:41:20 2018 daemon.info dnsmasq-dhcp[2320]: DHCPACK(br-lan) 10.1.1.11 20:a6:0c:3c:09:2d poco
Wed Dec  5 21:41:20 2018 daemon.info odhcpd[1515]: Using a RA lifetime of 0 seconds on br-lan
Wed Dec  5 21:41:21 2018 kern.info kernel: [ 2578.163231] device wlan1 left promiscuous mode
Wed Dec  5 21:41:21 2018 kern.info kernel: [ 2578.168150] br-lan: port 2(wlan1) entered disabled state
Wed Dec  5 21:41:21 2018 kern.info kernel: [ 2578.241745] device wlan0 left promiscuous mode
Wed Dec  5 21:41:21 2018 kern.info kernel: [ 2578.246397] br-lan: port 3(wlan0) entered disabled state
```

gwlim, 2018-12-05:
1) Is it all wireless client or only single unit? If Single Client check client for driver updates.

2) Did you overclock your hardware if so it might be too much, dial down on the overclock

3)Could be your Wireless Config file out of date? If so delete the wireless config file and start over
SSH in and do 
```
rm /etc/config/wireless
reboot
```

4) Could it be ART partition issue? If so try to flash another art partition but remember to backup beforehand.

I recently got a Archer C7v4 and I am using this image with no issue on Wireless Clients

demorphica, 2018-12-07:
Thank you very much for responding
This issue was happening for all clients
I tried to remove my ```/etc/config/wireless``` but that did not help

I was able to clear old files as follows & it worked 

```
umount /overlay && firstboot && reboot
```
both radios are stable now & i'm getting the benefits of your image

[image]

Thanks a lot for the pointers

closed 2018-12-07 by demorphica