gwlim/openwrt-sfe-flowoffload-ath79 #4: kernel modules issue installing kmod-hid

demorphica, 2018-12-07:
I use kmod-hid to connect to my APC UPS using a USB cable for which apcupsd needs to use kmod-hid

i see this error when i try to install 

```
 opkg install kmod-hid
Installing kmod-hid (4.14.82-1) to root...
Downloading http://downloads.openwrt.org/snapshots/targets/ar71xx/generic/packages/kmod-hid_4.14.82-1_mips_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-hid:
 *      kernel (= 4.14.82-1-f11667096b7c23e1dc9446b0665e3ac8) *         kernel (= 4.14.82-1-f11667096b7c23e1dc9446b0665e3ac8) *         kernel (= 4.14.82-1-f11667096b7c23e1dc9446b0665e3ac8) *
 * opkg_install_cmd: Cannot install package kmod-hid.
```

gwlim, 2018-12-07:
I can add kmod-hid next.

gwlim, 2018-12-10:
kmod-hid added

closed 2018-12-10 by gwlim