gwlim/openwrt-sfe-flowoffload-ath79 #6: Missing Cake Queuing Discipline

komawoyo, 2018-12-08:
Ty for the builds! 

WDR3600v1
JAN2019 firmware (normal - ie not mini)

missing cake queuing discipline, sqm scripts installed (sqm-scripts, luci-app-sqm)

[image]

gwlim, 2018-12-08:
Try go to SYSTEM- > STARTUP
ENABLE and START SQM
Then go to luci-sqm and choose CAKE

komawoyo, 2018-12-08:
Ty! this fixed it, oddly, I had to restart the service and reboot a few times to get it working. This is definitely on my end though.

closed 2018-12-09 by gwlim

Riotous, 2019-02-07:
@gwlim You may want to put this information in an FAQ because I was looking for the answer to this issue for a while until by chance I looked into these closed issues.

vicodin99, 2019-06-09:
Same as ghost here, @gwlim . Was about to report this as an issue until github suggested to look at this thread. It would be nice if this was added as part of the FAQ.

At least for me, I uninstalled luci-app-sqm and reinstalled it, then cake appeared in the drop down menu.