gwlim/openwrt-sfe-flowoffload-ath79 #7: Can u add some kmod for USB storages?

MadRock88, 2018-12-11:
kmod-usb-storage-extras
kmod-fs-ext4?

and need ohci support? 
when in core already ehci?

gwlim, 2018-12-12:
Next release

MadRock88, 2018-12-12:
thx, without these modification overlay section does not increase through a flash drive)

gwlim, 2018-12-12:
Note that I can't do this for small build only normal builds.
Just adding USB support in the kernel for small build will cause it to exceed 4MB

MadRock88, 2018-12-13:
it all for heavy weight modules + swap :)
on my 4300 not enough free space :_)
maybe it can be installed from opkg(special repo for it) , not include in firmware?
and if its need only me - survive without it)
now, is the fastest firmware on OPWRT in my opinion, and i dont want to slow it, or else)

gwlim, 2018-12-14:
Don't bother with swap it is slow, swap with a SATA harddisk is already slow.
swap with USB is even slower.
Optimize your required package to retain performance.
Added your requested kmods

mtarnovan, 2019-02-03:
@gwlim sorry to comment on a close issue, but I'm having trouble mounting a ext4 formatted, GTP partitioned usb disk. The disk is detected, I can see the following in `dmesg`:

```
[10602.750686] usb-storage 2-1:1.0: USB Mass Storage device detected
[10602.767688] scsi host0: usb-storage 2-1:1.0
[10603.804414] scsi 0:0:0:0: Direct-Access     WD       Elements 25A2    1004 PQ: 0 ANSI: 6
[10603.822366] sd 0:0:0:0: [sda] Spinning up disk...
[10604.859492] .
[10605.883484] .
[10606.907439] .
[10607.931435] .
[10608.955449] .
[10609.979417] .
[10609.981424] ready
[10609.985695] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[10610.000913] sd 0:0:0:0: [sda] 5860466688 512-byte logical blocks: (3.00 TB/2.73 TiB)
[10610.008852] sd 0:0:0:0: [sda] 4096-byte physical blocks
[10610.024469] sd 0:0:0:0: [sda] Write Protect is off
[10610.029379] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[10610.035702] sd 0:0:0:0: [sda] No Caching mode page found
[10610.041144] sd 0:0:0:0: [sda] Assuming drive cache: write through
[10610.109344]  sda: sda1
[10610.117005] sd 0:0:0:0: [sda] Attached SCSI disk
```

... but mounting fails with:
```
root@sibiu:/# mount -rt ext4 /dev/sda1 /data
mount: mounting /dev/sda1 on /data failed: Invalid argument
```
`fdisk` identifies the partitions correctly:
```
root@sibiu:/# fdisk -l /dev/sda
Disk /dev/sda: 2.7 TiB, 3000558944256 bytes, 5860466688 sectors
Disk model: Elements 25A2   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: gpt
Disk identifier: 3BD76470-BE45-4AB6-A304-87A7DAF4FE22

Device     Start        End    Sectors  Size Type
/dev/sda1   2048 5860466654 5860464607  2.7T Linux filesystem
```

Any ideas?

gwlim, 2019-03-04:
Try MAR 2019

MadRock88, 2019-03-04:
Maybe will work:/# mount -t -r ext4 /dev/sda1 /data

mtarnovan, 2019-03-04:
No, i tried. Will give the March release a try on the weekend

mtarnovan, 2019-04-16:
It works with MAR 2019, thanks @gwlim 

@MadRock88 doesn't matter, i was just showing that the mount fails.

closed 2019-04-18 by gwlim