gwlim/openwrt-sfe-flowoffload-ath79 #9: Publish changes?

oldium, 2018-12-19:
Would it be possible to publish used patches, so that I could for example build my own firmware based on yours (I usually do it based on the official openwrt git)? I would like to try the speed and compare it. I have two Archer C7 v4 and the speed offered by the official git sources is not satisfactory...

If it is really that fast, it would be good if it goes into official openwrt git (if not already being done).

Thanks.

gwlim, 2018-12-20:
Last time I did that didn't turn out well, I guess people are impossible to please.
You can still take the patches of the old repo.

closed 2018-12-20 by gwlim

dag03tsc, 2018-12-23:
Hi @gwlim !

I knee at you. Thanks a lot and congratulations for this awesome work! 💯 

I'm on the same boat as @oldium .

About a year or so I managed to compile my own v17.01.4 build for the WDR4900 but it required to move some patches to broken folder and to tweak some other ones. Despite that, I've been using it since then and it's been working flawlessly!

Now I'm in the need of adding some more packages and tweaking some conf but, while I'm on it, I would like to upgrade to v18.06.1 or, at least, v17.01.6.

I managed to achieve it with v17.01.6 thought it has required tweaking a couple of additional patches but I haven't tested the build on device yet.

I was about to do the same for v18.06.1 but it needs some more tweaking due to the recent inclussion of qca-ssdk and family, so I was hoping you could publish some repo with the whole updated bunch of patches and tweaks you apply on your **openwrt-sfe-flowoffload/JAN-2019** current working builds.

It'd be awesome if we could compile our own builds on top of it with all our package, tweak and conf needs and keep it updated from time to time.

If you rather not to make it full public, could you occasionally share it via PM or some other way?

Hoping you change your mind! :D

Thanks a lot in advance.

Greetings and happy holidays!

jack338c, 2019-01-11:
@oldium 
Did you build  gwlim's newest repo successfully on openwrt master?
I want to try his newone too
thanks!

oldium, 2019-01-11:
@jack338c I just created new DTS configuration for ath79, it was already working on ar71xx :-)

I just think it is beneficial to open-source your improvements on an open source project, so without up-to-date patches from @gwlim, I cannot know how cool and fast the world could be. But, anyway, this is life :-)

jack338c, 2019-01-11:
@oldium 
yeah,this is life :gift:

gwlim, 2019-01-11:
https://openwrt.org/docs/guide-user/perf_and_log/benchmark.nat

aleho, 2020-01-19:
You are publishing binary results of source code licensed under the GPL, therefor you have to publish the source code producing these binaries or provide a way to reproduce that code. All your changes made to source code licensed under the GPL will automatically be subject to that license as well (the GPL is viral). You may not change the licensing terms either.

You could comply with your obligations by e.g. providing a full source tree (fork) or patch files.

TLDR: You should release the source code to the firmware images in this repository in a format that is usable for anyone who wants to reproduce your changes.