gwlim/openwrt-sfe-flowoffload-ath79 #10: JAN 2019 -images (wdr4300_

asergiu, 2019-01-12:
Hi
Tested you January images - not sure is related to your image or to SFE but the upload images is badly impacted by the Flow offload.
I did a repeated test with and without SFE (check and service enabled and disabled and the results are 
always around these:
**Without Flow offlaod and SFE**:
[image]

**With flow offloading and SFE**:
[image]

AS you can see the download is benefitting from this but the upload goes down. My connection is **500Mb -down with 250 Mb/upload** roughly.

asergiu, 2019-01-12:
seems that without ticking flow offload - and just leaving the SFE service on - gets around 416Mb/s download and 240 Mb/s upload
[image]

elgab, 2019-01-14:
Yes, that is normal-ish.
for Digi gigabit PPPoE best performing implementation is SFE.
the overclocked 720Mhz WDR3600 is doing 800mbit down + full 500mbit up RDS PPPoE.

asergiu, 2019-01-14:
Yep - my connection is 500 down/250 Up max.
So SFE is around there.... 
you can close this.

closed 2019-01-14 by asergiu