gwlim/openwrt-sfe-flowoffload-ath79 #12: WNDR3700v1

robcore, 2019-01-14:
Hi there! 
Thanks for all of your hard work.  Would you be open to adding the Netgear WNDR3700v1 to supported devices?  It is similiar to the v4, but has a different cpu.  It is ar7xx, however, and i think would work well.
Take care!
Rob

gwlim, 2019-01-16:
Next release

robcore, 2019-01-16:
> Next release

Thanks so much, man!  Though if you're doing it just for me, you could save yourself the effort because I'm picking up a used V4 tomorrow from someone in my city (lucky, right?).

closed 2019-01-18 by robcore

gwlim, 2019-01-26:
Added support for WNDR3700v1

gwlim, 2019-01-29:
Added overclocked image for 800MHZ ar7161 based on overclock patch available online.
Please try ONLY if you know how to recover using uboot/redboot loader.
It will be a permanent thing only if I know it works