gwlim/openwrt-sfe-flowoffload-ath79 #13: sysctrl update required

insound, 2019-01-15:
I checked 11-nf-conntrack.conf in sysctl.d, and find thats net.netfilter.nf_conntrack_tcp_no_window_check=0 missed up. It shoud be turn off when sfe enabled. Thanks.

gwlim, 2019-01-16:
Hi I checked in a firmware with resetted config, there is no 
```
net.netfilter.nf_conntrack_tcp_no_window_check=0 
```
in 
/etc/sysctl.d/11-nf-conntrack.conf

Could it be you overwrote the file and preserved the setting when sysupgrade?

insound, 2019-01-17:
It was a clean install with factory reset. 
11-nf-conntrack.conf exist in your firmware, but without this line. I guess in sfe repo this line missed too) i read about this tcp parameter in sfe thread forum. Thx)

gwlim, 2019-01-26:
```
net.netfilter.nf_conntrack_tcp_no_window_check=1
```
is actually part of the SFE support code if you want to set it to 0 you should set it in sysctl

closed 2019-01-26 by gwlim