gwlim/openwrt-sfe-flowoffload-ath79 #16: kmod-usb-printer

azinchen, 2019-01-30:
Please add kmod-usb-printer

gwlim, 2019-03-03:
Added in MAR 2019

closed 2019-03-03 by gwlim

thomasrc, 2019-08-03:
Is it possible that kmod-usb-printer is missing from the latest build (AUG-2019)? I updated my Archer C7v2 and the mentioned package is not listed under the "installed" tab.

azinchen, 2019-08-03:
> Is it possible that kmod-usb-printer is missing from the latest build (AUG-2019)? I updated my Archer C7v2 and the mentioned package is not listed under the "installed" tab.

As I understand, all kmod- are removed from firmware, but compiled and distributed along with firmware. You can copy and install necessary kmods manually from kmod-package.tar.gz file.

thomasrc, 2019-08-03:
Ah, I see. Sorry, I missed that part of the changelog. :D