gwlim/openwrt-sfe-flowoffload-ath79 #18: Generating keys/certs for the Openvpn ?

silverus777, 2019-02-02:
Hello gwlim !

Thank you for the great work.
I'm newbie on Openwrt and sorry for a noob question.
I can't figure the method to generate the keys and cert on your version.
I read a lot of how-to's, but none of them seems to work.

I tried easy-rsa, build-ca, openssl but I've got only:

> -ash: openssl: not found
> -ash: easyrsa: not found
> -ash: build-ca: not found

I have packages installed that comes with your image:
libopenssl - 1.0.2p-1
openvpn-openssl - 2.4.6-3
wpad-openssl - 2018-05-21-62566bc2-7

openvpn-easy-rsa is shown as installed in Luci/System/Software but it's not listed with opkg list-installed.

I would greatly appreciate any pointer :)

silverus777, 2019-02-03:
Than you for the prompt answer.

Yes i want to make whole Openvpn config in the router.

Installing easyrsa was the first thing what i've tried.
However looks like it's impossiblile on this  build: :(
```

root@OpenWrt:~# opkg install openvpn-easy-rsa
Installing openvpn-easy-rsa (3.0.4-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/openvpn-easy-rsa_3.0.4-1_mips_24kc.ipk
Installing libopenssl1.0.0 (1.0.2q-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/libopenssl1.0.0_1.0.2q-2_mips_24kc.ipk
Collected errors:
 * check_data_file_clashes: Package libopenssl1.0.0 wants to install file /usr/lib/libcrypto.so.1.0.0
        But that file is already provided by package  * libopenssl
 * check_data_file_clashes: Package libopenssl1.0.0 wants to install file /usr/lib/libssl.so.1.0.0
        But that file is already provided by package  * libopenssl
 * opkg_install_cmd: Cannot install package openvpn-easy-rsa.
```

gwlim, 2019-02-03:
```
root@openwrt:~# opkg list-installed | grep easy-rsa
openvpn-easy-rsa - 3.0.4-1
```

silverus777, 2019-02-03:
Yes I think this might be a problem.
with
`opkg list-installed | grep easy-rsa
`
it is not listed !

However it is listed in Luci/Software/Installed !
also
```
root@OpenWrt:~# easyrsa
-ash: easyrsa: not found
root@OpenWrt:~#

```
Router is: WR1043ND V4

silverus777, 2019-02-03:
Is there sfe-flowoffload regular build you could recommend ?

gwlim, 2019-02-03:
I just ran this command on the NORMAL build MR3420v2
```

root@openwrt:~# opkg update && opkg install openvpn-easy-rsa
Downloading http://downloads.openwrt.org/snapshots/targets/ar71xx/tiny/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Installing openvpn-easy-rsa (3.0.4-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/openvpn-easy-rsa_3.0.4-1_mips_24kc.ipk
Installing openssl-util (1.0.2q-2) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/base/openssl-util_1.0.2q-2_mips_24kc.ipk
Configuring openssl-util.
Configuring openvpn-easy-rsa.
```

gwlim, 2019-02-03:
> Is there sfe-flowoffload regular build you could recommend ?

Which version did you download and flash?
The [latest FEB ones](https://github.com/gwlim/openwrt-sfe-flowoffload/tree/master/FEB-2019/openwrt-sfe-flowoffload-normal/TP-Link%20WR1043NDv4-mips74k) should have everything you need.

silverus777, 2019-02-03:
Easy-rsa installed finally ! Yes !
I used the flash from the link you've provided.
That was it, my fault, wrong version, sorry for the trouble.
And again, thank you for the quick and accurate help !

I'm closing the issue.

closed 2019-02-03 by silverus777