gwlim/openwrt-sfe-flowoffload-ath79 #19: Compatibility with mwan3

Adam1986, 2019-02-06:
Hello, I use a TP-Link 1043ND v2 and have a dual wan connection, so I use the mwan3 package.

If I was to switch to this image, would I expect to see any benefit on both connections?

gwlim, 2019-02-08:
I have patched the SFE module to marks in connections so it should work.
You can always backup your settings try it and restore if it doesn't work.
Let me know as well as I don't have a dual wan connection

Adam1986, 2019-02-08:
I had a very quick look this afternoon and did an upgrade retaining settings. Unfortunately I had to restore my backup image back because it doesn't look like there is enough space left to install the packages I need.  I have to use dnsmasq-full as I use ipsets on some of my mwan3 rules. 

I also use a xorpatched openvpn build that supports the scramble feature and I could see your build came with openvpn already. 

I did try removing a few packages but it didn't really give much space back.  I think I might have to check this out again in the near future to see if there's anything I can do.  Or would it be easy enough to create my own image with the bare essentials?

gwlim, 2019-02-09:
What scramble feature? If it is useful I can include it.
This firmware already comes with dnsmasq-full minus ipv6 as ipv6 on openwrt is provided by odhcpv6

Adam1986, 2019-02-10:
Here's a little more information on it:

https://tunnelblick.net/cOpenvpn_xorpatch.html

I've used it for a few years now as my mobile network provider at one point used to throttle all roaming traffic and this was a way around it. 

If it helps I can always provide you with the patched ipk files as I just sorted these out for the latest openwrt release ready to update my router at some point in the next week or two.

bkumar10, 2019-05-04:
I am not able to get mwan3 to work with this build. Actually, I am unable to get two interfaces working at the same time. They show up in ip route. A ping to default gateway works fine but a ping to an external address does not work.

gwlim, 2019-08-03:
Changed the way SFE determine the destination path. You can try again.

Feche, 2019-09-15:
I can confirm that _mwan3_ doesn't work with this version of OpenWRT

Debug:

Version: OpenWrt SFE r11009-1cf2495d48 / LuCI ( Trunk )
mwan version: 2.8.0-2

iptables-save output:
`
[...]
# Completed on Sun Sep 15 13:55:00 2019
# Generated by iptables-save v1.8.3 on Sun Sep 15 13:55:00 2019
*mangle
:PREROUTING ACCEPT [60998:75091829]
:INPUT ACCEPT [1802:116501]
:FORWARD ACCEPT [59111:74968984]
:OUTPUT ACCEPT [2341:3671257]
:POSTROUTING ACCEPT [61452:78640241]
:mwan3_connected - [0:0]
:mwan3_hook - [0:0]
:mwan3_iface_in_wan - [0:0]
:mwan3_iface_in_wan2 - [0:0]
:mwan3_ifaces_in - [0:0]
:mwan3_policy_balanced - [0:0]
:mwan3_policy_wan2_only - [0:0]
:mwan3_policy_wan_only - [0:0]
:mwan3_rules - [0:0]
-A PREROUTING -j mwan3_hook
-A FORWARD -o wlan1 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A OUTPUT -j mwan3_hook
-A mwan3_connected -m set --match-set mwan3_connected dst -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_hook -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
-A mwan3_iface_in_wan -i wlan1 -m set --match-set mwan3_connected src -m mark --mark 0x0/0x3f00 -m comment --comment default -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_iface_in_wan -i wlan1 -m mark --mark 0x0/0x3f00 -m comment --comment wan -j MARK --set-xmark 0x100/0x3f00
-A mwan3_iface_in_wan2 -i eth0.2 -m set --match-set mwan3_connected src -m mark --mark 0x0/0x3f00 -m comment --comment default -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_iface_in_wan2 -i eth0.2 -m mark --mark 0x0/0x3f00 -m comment --comment wan2 -j MARK --set-xmark 0x200/0x3f00
-A mwan3_ifaces_in -m mark --mark 0x0/0x3f00 -j mwan3_iface_in_wan
-A mwan3_ifaces_in -m mark --mark 0x0/0x3f00 -j mwan3_iface_in_wan2
-A mwan3_policy_balanced -m mark --mark 0x0/0x3f00 -m statistic --mode random --probability 0.50000000000 -m comment --comment "wan2 1 2" -j MARK --set-xmark 0x200/0x3f00
-A mwan3_policy_balanced -m mark --mark 0x0/0x3f00 -m comment --comment "wan 1 1" -j MARK --set-xmark 0x100/0x3f00
-A mwan3_policy_wan2_only -m mark --mark 0x0/0x3f00 -m comment --comment "wan2 1 1" -j MARK --set-xmark 0x200/0x3f00
-A mwan3_policy_wan_only -m mark --mark 0x0/0x3f00 -m comment --comment "wan 1 1" -j MARK --set-xmark 0x100/0x3f00
-A mwan3_rules -s 192.168.1.0/24 -m mark --mark 0x0/0x3f00 -m comment --comment lan -j mwan3_policy_wan_only
-A mwan3_rules -s 10.10.77.0/24 -m mark --mark 0x0/0x3f00 -m comment --comment lan2 -j mwan3_policy_wan2_only
COMMIT
[...]
´

ip route output:

`default via 192.168.0.1 dev wlan1 proto static src 192.168.0.200
10.10.77.0/24 dev eth0.3 proto kernel scope link src 10.10.77.254
190.210.xxx.0/24 dev eth0.2 proto kernel scope link src 190.210.xxx.xxx
192.168.0.0/24 dev wlan1 proto kernel scope link src 192.168.0.200
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1`

Feche, 2019-09-15:
I also tried mwan3 2.6.18-1 and this is the **iptables-save** output:

>*mangle
:PREROUTING ACCEPT [66642:81396568]
:INPUT ACCEPT [632:61222]
:FORWARD ACCEPT [65929:81323023]
:OUTPUT ACCEPT [598:116677]
:POSTROUTING ACCEPT [66481:81436726]
:mwan3_connected - [0:0]
:mwan3_hook - [0:0]
:mwan3_iface_in_wan - [0:0]
:mwan3_iface_in_wan2 - [0:0]
:mwan3_iface_out_wan - [0:0]
:mwan3_iface_out_wan2 - [0:0]
:mwan3_ifaces_in - [0:0]
:mwan3_ifaces_out - [0:0]
:mwan3_policy_balanced - [0:0]
:mwan3_policy_wan2_only - [0:0]
:mwan3_policy_wan_only - [0:0]
:mwan3_rules - [0:0]
-A PREROUTING -j mwan3_hook
-A FORWARD -o wlan1 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A OUTPUT -j mwan3_hook
-A mwan3_connected -m set --match-set mwan3_connected dst -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_hook -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_out
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
-A mwan3_iface_in_wan -i wlan1 -m set --match-set mwan3_connected src -m mark --mark 0x0/0x3f00 -m comment --comment default -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_iface_in_wan -i wlan1 -m mark --mark 0x0/0x3f00 -m comment --comment wan -j MARK --set-xmark 0x100/0x3f00
-A mwan3_iface_in_wan2 -i eth0.2 -m set --match-set mwan3_connected src -m mark --mark 0x0/0x3f00 -m comment --comment default -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_iface_in_wan2 -i eth0.2 -m mark --mark 0x0/0x3f00 -m comment --comment wan2 -j MARK --set-xmark 0x200/0x3f00
-A mwan3_iface_out_wan -o wlan1 -m mark --mark 0x0/0x3f00 -m comment --comment wan -j MARK --set-xmark 0x100/0x3f00
-A mwan3_iface_out_wan2 -o eth0.2 -m mark --mark 0x0/0x3f00 -m comment --comment wan2 -j MARK --set-xmark 0x200/0x3f00
-A mwan3_ifaces_in -m mark --mark 0x0/0x3f00 -j mwan3_iface_in_wan2
-A mwan3_ifaces_in -m mark --mark 0x0/0x3f00 -j mwan3_iface_in_wan
-A mwan3_ifaces_out -m mark --mark 0x0/0x3f00 -j mwan3_iface_out_wan2
-A mwan3_ifaces_out -m mark --mark 0x0/0x3f00 -j mwan3_iface_out_wan
-A mwan3_policy_balanced -m mark --mark 0x0/0x3f00 -m statistic --mode random --probability 0.50000000000 -m comment --comment "wan2 1 2" -j MARK --set-xmark 0x200/0x3f00
-A mwan3_policy_balanced -m mark --mark 0x0/0x3f00 -m comment --comment "wan 1 1" -j MARK --set-xmark 0x100/0x3f00
-A mwan3_policy_wan2_only -m mark --mark 0x0/0x3f00 -m comment --comment "wan2 1 1" -j MARK --set-xmark 0x200/0x3f00
-A mwan3_policy_wan_only -m mark --mark 0x0/0x3f00 -m comment --comment "wan 1 1" -j MARK --set-xmark 0x100/0x3f00
-A mwan3_rules -s 192.168.1.0/24 -m mark --mark 0x0/0x3f00 -m comment --comment lan -j mwan3_policy_wan_only
-A mwan3_rules -s 10.10.77.0/24 -m mark --mark 0x0/0x3f00 -m comment --comment lan2 -j mwan3_policy_wan2_only
COMMIT

As you can see, this version has the chains _mwan3_iface_out_wan_, _mwan3_iface_out_wan2_ and _mwan3_ifaces_out_ , so there may be actually more than one problem with this version of OpenWRT, 1st is that the mwan3 package is broken and 2nd that the router doesn't handle the MARKs set by _mwan3_ but since mwan3 version 2.6.18-1 adds all the correct chains, the issue is on the ROM.

If you need anything else, I am pleased to help since I need fast path and mwan working!
Thanks

gwlim, 2019-09-16:
Try the rework version I added mark parsing

Feche, 2019-09-16:
Yes, it was tested with SEP version and it is not working.

mrfixit2001, 2020-12-04:
I just tested in my Archer C7 V2. Something is not allowing both radios to work as a client at the same time. When I have either the 5ghz or the 2.4ghz active, with the other radio disabled, I can successfully ping -I out the active interface. If I have both radios active at the same time then only the ping -I works for the 5ghz interface, the 2.4 interface has 100% loss. This doesn't seem to be happening on the current regular release of openwrt. It should very possibly be routing related. Any advice? I have disabled mwan3 but left the installation in place so I can perform any tests you'd recommend to try and resolve.

mrfixit2001, 2020-12-07:
Reverting back to the dev builds of openwrt master resolves, so I'll stick with that until this is addressed.