gwlim/openwrt-sfe-flowoffload-ath79 #20: Some benchmarks with RB450G

eisengrau, 2019-02-09:
Hello,

Not exactly issues, I just did some jperf benchmarks with my RB450G. Basically the below images speak to themselves:
[image]

The above is with sortcut-fe enabled. flowoffload will yield similar results. Enabling both will degrade the performance a bit, but not as much as below:

So this is WITHOUT any of sortcut-fe or flowoffload enabled:
[image]

The router was running OpenWrt SFE r8810-09004e6e13 (the january release), and with the stock 680Mhz clock. Overclocking it to 800Mhz will possible allow it to go beyond 1000, but that I haven't tried... So, as you can see, using gwlim's images, the RB450G will perfectly handle consumer Gigabit WAN (as we all know, residential fibre gigabit will never achieve that speed - an ISPs guaranteed link speed is 0kbps)!

gwlim, 2019-02-09:
Nice! 👍 
Thanks for the feedback!

closed 2019-02-09 by gwlim

gwlim, 2019-02-12:
To add on to test effects of SFE and  SQM
1)To test SFE, turn on SFE enable and start
Test jperf > 900Mbps
2)To test Traffic passthough QDISC even when SFE is enabled, turn on SQL Cap Rate Limit to 300Mbps
Notice that jperf < 300Mbps
3)To test maximium possible SQM with SFE, set SQM to 900Mbps
Notice that jperf graph is fluctuating, jperf < 400Mbps
Therefore maximium performance with SFE and SQM  = < 400Mbps