gwlim/openwrt-sfe-flowoffload-ath79 #21: SQM: The installed version of package kernel is not compatible, require 4.14.98-1-6da5e890… while 4.14.95-1-327182cc… is installed.

makruiten, 2019-02-09:
I have a dependency problem when I try to install SQM on a 32MB TL-WR1043ND with the FEB-2019 build installed. Can I install it anyway?

makruiten, 2019-02-09:
Wouldn't it be better to build with the same kernel as all the package are built against?

Riotous, 2019-02-09:
@makruiten Isn't it just because gwlim's builds are running a newer kernel?

makruiten, 2019-02-09:
Yes and no. It’s a newer kernel than what is used for these devices in the official builds, but older than the kernel used for some other devices. In this case opkg complains that the kernel is older than the one that is required by the package. It looks to me like this could be fixed by using the same (newer) kernel version, but that’s just an assumption.

makruiten, 2019-02-09:
Or maybe it is the same kernel: https://openwrt.org/releases/18.06/notes-18.06.2. But then I don’t understand why the package requires a newer kernel than the one installed.

gwlim, 2019-02-11:
What is the name of the package that triggered this message?
I will try to add that the mini builds are really quite full just adding 1 package would cause it not to build.

makruiten, 2019-02-11:
It was luci-app-sqm, so I guess you left it out exactly for that reason :)

Whenever you build a new image, do you then use the same kernel as the packages in the [current snapshot](https://downloads.openwrt.org/snapshots/packages/mips_24kc/)? If that's the case, I can wait for your next build and then install SQM while it's fresh. I'm not in a hurry!

gwlim, 2019-02-12:
The thing is luci-app-sqm should not have any dependency on the linux kernel, it pulls some other packages that depend on the kernel. kmod-something

gwlim, 2019-02-12:
Sorry accidentally close the issue

makruiten, 2019-02-12:
I see! I can check that, but right now the router is in use with an official build on it, so I'll have to check that later. Judging from the current installed packages, that would probably include kmod-sched-cake and kmod-sched-core.

gwlim, 2019-03-03:
I checked dependency includes kmod-ifb and kmod-sched-core, modules added.
Reopen if there are still missing dependencies.

makruiten, 2019-03-03:
I can't get it to install:

[image]

```
root@TL-WR1043ND:~# opkg install luci-app-sqm
Installing luci-app-sqm (1.3.0-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/luci-app-sqm_1.3.0-1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-sqm:
 * 	kernel (= 4.14.103-1-7b96d9ca0358834b880712b6d716f275)
 * opkg_install_cmd: Cannot install package luci-app-sqm.
```

gwlim, 2019-03-04:
Added Kernel modules added based on the dependency tree.
Please try again

makruiten, 2019-03-04:
I think that you've put the image it in the normal directory instead of the mini directory, but it's still the mini build, right? I'll test it.

makruiten, 2019-03-04:
Installation now succeeds, thanks a lot! I'll leave the issue open, since it's now in the wrong directory of the repo.

gwlim, 2019-03-04:
Bummer put the files in wrong directory, thanks for the heads up.

closed 2019-03-04 by gwlim