gwlim/openwrt-sfe-flowoffload-ath79 #26: Switch from ar71xx to ath79 for routers like Archer C7?

Riotous, 2019-02-24:
https://downloads.openwrt.org/snapshots/targets/ath79/generic/openwrt-ath79-generic-tplink_archer-c7-v2-squashfs-factory.bin

Apparently, ath79 snapshots will be the new default for Atheros based routers like the Archer C7 so will these builds also move to that?

gwlim, 2019-03-03:
Will switch when the time comes, in terms of code updates it is the same, just without the DTS boot feature which has no effects on functionality, only during boot.

MOZGIII, 2019-03-13:
What's DTS boot feature?

gwlim, 2019-03-14:
DTS=Device Tree, the file extension is .dts so I usually call it DTS

MOZGIII, 2019-03-14:
Oh, got it!
Btw, it seems to be solved with March release: https://github.com/gwlim/openwrt-sfe-flowoffload/blob/master/MAR-2019/openwrt-sfe-flowoffload-normal/TP-Link%20Archer%20C7v2-mips74k/openwrt-ar71xx-generic-archer-c7-v2-squashfs-factory.bin

We can close this issue  😊

closed 2019-03-19 by gwlim