gwlim/openwrt-sfe-flowoffload-ath79 #32: Question: does this help speed up all connections or WAN?

ShapeShifter499, 2019-04-02:
I have a supported router, Western Digital n750 and currently run LEDE 17.01   I'm planning on updating to 18.06 but I'm not sure which is better, these builds or the official builds.

I want whatever build that is likely to give the best LAN speed. My ISP speed is only 15Mbps down and 5Mbps up, so I don't think anything will help much there.

MadRock88, 2019-04-03:
As far as I understand, your ISP gives you only 15 MB, so you don’t need super optimizations to connect LAN-WAN.
The speed inside the network between the internal computers (LAN-LAN) is limited by a switch, and they work without any loss in speeds in most cases. That is, the computers themselves are limiters.

Therefore, the firmware should be chosen from personal preferences, which is necessary from the "box"

ShapeShifter499, 2019-04-04:
Just to be clear. The patches and features of this firmware

closed 2019-04-04 by ShapeShifter499

ShapeShifter499, 2019-04-04:
I'll close this but I just want to make sure I'm understanding some stuff.

@MadRock88 if for whatever reason I'm still using this hardware in the future and get faster internet. The patches and scripts in this firmware might help. But since my WAN-LAN is far below Gigabit speeds I don't really have to worry much. 

Same goes if I assign the WAN port as just another LAN port since I'm not typically using NAT anymore.

Correct?