gwlim/openwrt-sfe-flowoffload-ath79 #33: 4/5 of the IOT devices (esp8266/esp32) can not connect to the WIFI AP

kbhuinfo, 2019-04-08:
Upgrade from 14.07 succeeded and the settings were applied manually. The configs are the same in the access point, but only a few devices can connect to it. How could this happen? I was playing around with the settings, but unfortunately, no success.

Main question: what has changed since 14.07 regarding WIFI?

(everything works fine and stable except for this issue)

MadRock88, 2019-04-08:
same issue with HP wifi printer
on stock firmware - connected normaly, 
on OWRT since 17.04 (Lede) and now that release  - very very long connection time. IDK why.
i try 20/40 wight, channels from 1 to end - no reactions. 
any smartphones - work without troubles.
by wired connection - all fine.

kbhuinfo, 2019-04-08:
Thanks for the reply and the info. I tried 2.4GHz freq. and "N" and "Legacy" settings, 20MHz bandwith. No success. Always the same 3-4 devices are connecting, the rest (~20) can not. I am not sure, where the connection brakes, I could not analyse it further on.

MadRock88, 2019-04-08:
1 thing I not try, encryption metod, ALL my attempts was be on WPA2&WPA mixed, WPA, WPA2 mode
not try open wifi or WEP

kbhuinfo, 2019-04-10:
The workaround, to let a separate AP handle the wifi clients worked - but the issue still exists. Can someone help? What can we do to support the error correction?

gwlim, 2019-04-11:
Can you try removing /etc/config/wireless and start over?

kbhuinfo, 2019-04-11:
> Can you try removing /etc/config/wireless and start over?

I can do it today evening! (CET) Thanks in advance!

kbhuinfo, 2019-04-11:
Before "uci commit" I removed the APs manually and erased /etc/config/wireless. After it, I rebooted the router and started again from scratch. Same result, unfortunately.

kbhuinfo, 2019-04-16:
Since then, I tried other channels - without success. How can I help you to go on with the analysis/resolution?

gwlim, 2019-04-16:
Could it be due to Anti-KRACK vulnerability prevention?

kbhuinfo, 2019-04-16:
Thanks for the hint - I will check it tonight!

kbhuinfo, 2019-04-16:
Krack not enabled and all other options are disabled in wireless security as well. Security is "WPA-PSK/WPA-PSK2 Mixed Mode" and cipher is "auto", as before. Only "Short Preamble" and "Disassociate On Low Acknowledgement" are set in advanced settings.

Pentaphon, 2019-04-22:
>Security is "WPA-PSK/WPA-PSK2 Mixed Mode" and cipher is "auto", as before. 

@kbhuinfo Do WPA2 only and AES cipher only. That's usually recommended, security-wise.

kbhuinfo, 2019-04-23:
Will give a try, thank you!

closed 2019-08-03 by gwlim

filipsPL, 2021-02-18:
Just for future reference: I had the same problem. For me reloading the wifi configuration helped. I changed country code settings in the luci interface for one of the network, applied, and after the service restart everything started working.