gwlim/openwrt-sfe-flowoffload-ath79 #41: Baby jumbo frames no longer supported?

timdodge, 2019-05-19:
Hi,

Thanks for providing these builds, they're really good thanks.

One thing I've noticed is that moving from a firmware built from openwrt-18.06, we've lost the ability to set the MTU of the wan interfaces to 1508, so that the pppoe interface gets a MTU of 1500.

I have the following in network:
`config interface 'eth0'
        option ifname 'eth0'
        option mtu '1508'`

But this results in the following kernel message:
`[   28.387018] eth0: Invalid MTU 1508 requested, hw max 1500`

As a result the pppoe-wan interface ends up with a MTU of 1492, which isn't ideal.

Any idea what has changed between 18.06 and trunk?

Thanks,
Tim

Dean-Bayliss, 2019-06-07:
**This build is excellent**
I appear to have the same issue as Tim. On the PPPoE interface, I used to bump the MTU up, to get the full 1500 byte packets (RFC 4638).  I set the MTU option to 1508 it would then intelligently negotiate with the ISP's server for a 1500 MTU. The last build I was using was LEDE Reboot 17.01.4 - on WDR4900v1 and that has worked fine and previous versions.

Currently pppoe-wan reports  MTU:1492 - please let me know if I can help with any testing or do I need to adjust my configuration?  Thanks.

Dean

gwlim, 2019-06-08:
Can you try upstream OpenWRT snapshots and see if that changed the MTU?
Need to isolate the cause of change, if it happens on OpenWRT upstream then the changes might have propagated from there.

Dean-Bayliss, 2019-06-08:
Hi 
http://downloads.openwrt.org/snapshots/targets/mpc85xx/generic/
"At this moment snapshot images of WDR-4900 are broken.  TP-Link TL-WDR 4900 uses u-boot with read-only env.  The boot command reads only 0x29F000 data from flash.  Bigger images causes crc error"!
**What I'll do today is test the 18.06.0x versions as I wasn't sure if Tim had the issue on all versions and report back.**

timdodge, 2019-06-08:
I've moved back to to a self-built 18.02 for the moment:
```Model TP-Link Archer C7 v2
Architecture Qualcomm Atheros QCA9558 ver 1 rev 0
Firmware Version OpenWrt 18.06-SNAPSHOT r7778-5d27e87de7 / LuCI openwrt-18.06 branch (git-19.156.63894-115c4e3)
Kernel Version 4.9.180
```
I can confirm that I'm able to set the MTU of eth0 to 1508.

Dean-Bayliss, 2019-06-08:
My results were not conclusive, 18.06 not working either (which may be good news..) my issue is either ISP related or I need go back further to check with 15.05.01 where I have the log with the ping -D -s 1472 xxxx showing it working...

gwlim, 2019-06-09:
> 
> 
> I've moved back to to a self-built 18.02 for the moment:
> 
> ```
> Architecture Qualcomm Atheros QCA9558 ver 1 rev 0
> Firmware Version OpenWrt 18.06-SNAPSHOT r7778-5d27e87de7 / LuCI openwrt-18.06 branch (git-19.156.63894-115c4e3)
> Kernel Version 4.9.180
> ```
> 
> I can confirm that I'm able to set the MTU of eth0 to 1508.

Use snapshot to test, all the releases are based on snapshot it is the kernel version changes that might be the issue kernel version 4.14

Dean-Bayliss, 2019-06-09:
Just one last comment from me - test on 15.05.01/3.18.23 - and no change - For me this means my ISP is no longer supporting RFC 4638 MSS and not specifically related to the current development.

Dean-Bayliss, 2019-06-12:
I could not leave it alone.  Testing again on 15.05.01 - clean boot and not connected to the fibre modem, and ifconfig reports eth0 at 1508 mtu immediately as per config.  Connecting to the modem and ethernet connection remains at 1500 rather than 1492.  So old behaviour and fine.  No explanation as to why works as all that is different is a few days on....  (I have two of these routers - different firmware on each).

Disconnect from the modem.  Clean boot on **MARCH release SFE** and eth0 is set at 1500 mtu despite being told in the config to be 1508.  So I then manually set ifconfig eth0 mtu 1508 and no complaints it sets to 1508.  Connection then made to the modem and is fine and MAX PLAYLOAD negotiated - as shown 0x5DC = 1500 negotiated.  So now on **SFE-MARCH build I am connected no issues, using baby jumbo frames**.  There is just something odd, about the timing of that set up.  I wouldn't say this is conclusive, but clearly working.

================================================================
eth0      UP BROADCAST RUNNING MULTICAST  **MTU:1508**  Metric:1
pppoe-wan UP POINTOPOINT RUNNING NOARP MULTICAST  **MTU:1500**  Metric:1

daemon.info pppd[9548]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
daemon.debug pppd[9548]: **Send PPPOE Discovery** V1T1 PADI session 0x0 length 27
daemon.debug pppd[9548]:  [service-name] [host-uniq  36 30 ] [**PPP-max-payload  05 dc**]
daemon.debug pppd[9548]: **Recv PPPOE Discovery** V1T1 PADO session 0x0 length 64
daemon.debug pppd[9548]:  [AC-name BKLN05-BNG-01] [host-uniq  36 30 3] [**PPP-max-payload  05 dc]** [service-name] [AC-cookie  cb 01 b8 c5 a0 11 79 1b ae 31 96 12 fc 23 1f b3]

vicodin99, 2019-06-13:
So, March 2019 trunk still had baby jumbo frames working? What is the revision of the build?

That narrows it down quite a bit, as to find the offending commit...

Dean-Bayliss, 2019-06-15:
On mpc85xx/WDR4900v1 - **All SFE versions can be set to 1508**.  So **no issues on that platform**.  [Detail below].  Tim is using different hardware and maybe the issue he is seeing is specific to the platform being **Qualcomm Atheros QCA9558 ver 1 rev 0**

Working 
Model	TP-Link TL-WDR4900 v1, Processor Freescale P1014, Architecture e500v2, Firmware Version **OpenWrt SFE r9508-c1a8054114 / LuCI (Trunk)**
**OpenWrt SFE r10115-593b487538 / LuCI ( Trunk ), Kernel Version	4.14.121**

closed 2019-08-12 by gwlim