gwlim/openwrt-sfe-flowoffload-ath79 #42: JUN-2019 Download velocity degradation

xchwarze, 2019-05-26:
Hardware: Archer C7 V5

After JUN-2019 version download vel is always > 100Mbps
But after upgrade vel < 92Mbps

speedtest results: http://i67.tinypic.com/ziln4x.jpg
JUN-2019 VERSION
----------------------
5/26/2019 5:49 PM 
5/26/2019 4:50 PM

MAR-2019 VERSION
----------------------
5/26/2019 5:55 PM 
5/21/2019 10:51 PM 
3/10/2019 8:57 PM
3/10/2019 7:10 PM

gwlim, 2019-05-26:
Check Network -> Switch
Make sure link rate has not drop to 100Mbps, if it has reseat the cable.
Logically speaking even without SFE router is able to exceed 120Mbps easily.

xchwarze, 2019-05-30:
> 
> 
> Is the test using wireless or wired?
> Check Network -> Switch
> Make sure link rate has not drop to 100Mbps, if it has reseat the cable.
> Logically speaking even without SFE router is able to exceed 120Mbps easily.

Wired
1000Mbps

After reflash JUN-2019 version its ok 
download vel is always > 100Mbps again

closed 2019-05-30 by xchwarze