gwlim/openwrt-sfe-flowoffload-ath79 #43: JUN-2019 Mikrotik RB450G image unbootable / missing mini builds(?) / no 5G wifi

eisengrau, 2019-05-29:
Hello Again,

First of all I would like to thank your efforts, also in the name of the community, very much appreciate it!

I have tested your latest builds, and I encountered the below issues:

- Mikrotik RB450G not booting after update. Dug out my trusty null-modem cable to check the console, it is stuck at loading kernel. Could be because by default it is overclocked? I actually reverted to the MAR-2019 build, but this time I tried the overclocked *-oc* named image, and I got the same result: stuck at loading kernel. I then tried to flash the non -oc MAR image, and this time it booted properly. Since this device can be overclocked to 800Mhz max from the pre boot console interrupt menu (never tried though - may need hardware modifications, eg. bigger heat sinks on the chips. Also not really necessary if one does not have at least >= 1Gbit WAN), is it really necessary ti release a pre-overclocked image?

- TP-Link WDR3600 - 5GHZ Radio does not seem to work. It is detected, initialized. I have set a different SSID, WPA2 AES encryption ON, channel auto, transmit power auto. No go, devices will not detect it. 2.5Ghz works OK. With the MAR-2019 build I had both radios working. Not sure what could be the issue. Will revert to MAR and let you know.

- This is trivial, but I was not able to locate the mini builds in the repository. I was looking for 1043ND v1 32MB. You mentioned you got Wireguard in it, and I wanted to try it.

Thank you!

gwlim, 2019-05-30:
1) Yes that was clocked to 800MHZ, since no one commented that the OC image was not working I thought it was. So do you know what is the max stable overclock of AR7161? 750MHZ?
It is a old device so it should be OCed to maximise the performance. I can rebuild this weekend

2)I am quite confident WDR3600 is working because I tested it on my WDR4300 and 5GHZ works.
If you click on scan and you see 5GHZ Access Points means Wireless is working but there is something wrong with your /etc/config/wireless

3)Check the tar.gz in the folder all kmod-wireguard should be inside

eisengrau, 2019-05-30:
Hello,

Indeed, you can scratch out the WiFi issue with wdr3600, it was my fault, I accidentally disabled the ucitrack service and for that reason it did not apply my configuration for the 5GHz radio. sorry for that, working fine now!

Basically I haven't tried overclocking the rb450g yet, in the menu there is only a possibility to over/under clock it to predefined freqs. For example the next and last step is to go to 800Mhz from the stock 680Mhz. That's it, there is no possiblility to go 750. I'll try this next time when I'll remove it from the cabinet.

Oh, I see. Is it safe to flash this:

https://github.com/gwlim/openwrt-sfe-flowoffload/tree/master/JUN-2019/openwrt-sfe-flowoffload-normal/TP-Link%20WR1043NDv1-64MB-RAM-mips24k-ath9k 

on the 32MB umodded variant too? Then I can grab and install whichever kmod ipk I need from the archive.

Thank you,

gwlim, 2019-05-30:
Nope sorry realised I missed the mini builds will do it soon

eisengrau, 2019-05-31:
Awesome! Thank you! Waiting to update all my devices throughout the estate... :)

eisengrau, 2019-06-01:
Hai, what happened here? I was just in the middle of downloading and flashing the mini build for the 1043nd 32MB, but the next moment the image was gone from the folder. Have you updated something?

gwlim, 2019-06-01:
yes I just completed it.

eisengrau, 2019-06-01:
Yes! It's ok now, I was able to flash your latest image onto the "ultimate gigabit router" 1043nd 32mb beast! Now, I can be in the same "LAN" with my mate from the other butthole of Europe, thanks to the super-awesome wireguard (he is using the same 1043nd since years... and he was skeptical to reflash from stock tp-lin software, until now -yikes! -).

Thank you, again, gwlim, for your continous efforts. It's a breath of fresh air for these 'legacy' devices, and we definately need this kind of a "fresh air". :)

gwlim, 2019-06-02:
No problem. Enjoy.

closed 2019-06-02 by gwlim