gwlim/openwrt-sfe-flowoffload-ath79 #44: Adding Meraki MR16

Hurricos, 2019-05-31:
These have an AR7161 with 16F / 64R, and use the GBe coming out of the SoC itself. I would happily supply you with two of these routers, for the purposes of testing - assuming you would need that.

Note that there will be some delay before I can get them flashed with normal OpenWrt if you wanted me to poke around and test things myself - waiting for a a USB to 3v3 serial cable to come in the mail.

Hurricos, 2019-06-01:
Correction - they do not use the on-board GBe. They use an AR8021-BL1E. Do you reckon it would still be possible to flow offload?

gwlim, 2019-06-01:
Added firmware

closed 2019-06-01 by gwlim

Hurricos, 2019-06-08:
@gwlim Do you think you could either recompile this build with overclocking @ 800, or give me a link to your build system?

Thanks!

gwlim, 2019-06-12:
Someone commented that overclock isn't working for AR7161 on Mikrotik Routers.