gwlim/openwrt-sfe-flowoffload-ath79 #46: K2T out of memory and load is too high

yy0c, 2019-06-15:
[image]
Luci and SSH is responsive only for less than about 30 seconds, I've not changed any setting. I have no idea what's happening, would you please help me investigate this bug?

gwlim, 2019-06-15:
I can't see the full logs.
Copy and paste it here

yy0c, 2019-06-15:
> 
> I can't see the full logs.
> Copy and paste it here

Which ones do you need me to send?

gwlim, 2019-06-15:
The whole top log which processes are taking how much RAM

yy0c, 2019-06-16:
The weird thing is,  after waited for a minute or two, the system would be responsive again.  I guess we can close this issue for now, if you agree.

gwlim, 2019-06-16:
Sure if it works great :)

closed 2019-06-16 by gwlim