gwlim/openwrt-sfe-flowoffload-ath79 #49: Please keep update your firmware!

Zsolteeca, 2019-07-05:
The June version is really outdated :( 
Please give a newer version of components because some of the software elements are unusable.

Thank you!

gwlim, 2019-07-08:
Yep I understand there is a lot of big change from upstream
Next release is almost ready but I need to fix a couple of bugs I discovered especially when it impacts performance.

gwlim, 2019-07-08:
Looks like they did a large scaled change on LuCI again

Zsolteeca, 2019-07-08:
What kind of changes did they "commit"?

gwlim, 2019-07-08:
From Lua render of views to javascript render of views

Zsolteeca, 2019-07-09:
I am looking forward to a new version of fastpath lede

Zsolteeca, 2019-07-12:
When will it be available?

gwlim, 2019-07-12:
There is a serious bug I discovered.
https://bugs.openwrt.org/index.php?do=details&task_id=2364&order=dateopened&sort=desc
If your router gets a flood of UDP traffic your router can run out of memory and crash, upstream issue with kernel 4.19

Zsolteeca, 2019-07-12:
:(
You can always find bugs.. Every release of openwrt contains bugs..
I am waiting for a Fastpath Lede new actual release..
Probably a bug fix could be released for errors. You have a module attended system updates 
attendedsysupgrade-common and will it be work? Because I have a message that my router isn't supported..

gwlim, 2019-07-15:
I sort of understood the bug and alleviated the issue on WR1043NDv1.
Now I am testing it on my network.
I am going to dump http-dns-proxy because the author made some changes and cause unaligned memory access on MIPS cpu.

Pentaphon, 2019-07-22:
@gwlim Thanks for keeping this firmware updated. I look forward to your next release.

gwlim, 2019-07-22:
Released AUG

closed 2019-07-22 by gwlim