gwlim/openwrt-sfe-flowoffload-ath79 #53: Slow ipv6 with pppoe

kolomparrudi, 2019-07-21:
Hi,

My ISP (Digi Hungary) offers ipv6 in shared session with ipv4.
I have a TL-WR1043N/ND v3, with your June image, ipv4 benchmark results are 300-350 Mbps, ipv6 is around 100 Mbps. There is no difference in ipv6 speed if I stop shortcut-fe service.

Also have some questions about sfe: What is the difference between your fastpath sfe solution and the flow offloading offered by the latest openwrt snapshots? Do you have benchmarks which solution performs better? Do you think using real hardware offloading with AR8327n we can reach higher throughput?

Thanks,
Kolomparrudi

gwlim, 2019-07-22:
Do cat /sys/fast_classifier/debug_info to see if IPv6 is offloaded

gwlim, 2019-07-22:
Can also be due to bandwidth limitation set by website?

kolomparrudi, 2019-07-22:
Hi,

Checked the /sys/fast_classifier/debug_info, but only ipv4 connections present here. It seems my ipv6 connections are not offloaded.

I have already tested the bandwidth with multiple sites and with iperf3 and wget download too.

gwlim, 2019-07-22:
When I tested IPv6 static routing I can see the offload but not sure why it doesn't work in actual deployments. I need to troubleshoot this.

kolomparrudi, 2019-07-22:
If you need logs/debug/existing config/etc I'll try to help you.

gwlim, 2019-08-03:
Issue should be fixed tested with IPv6 enabled ISP

closed 2019-08-03 by gwlim