gwlim/openwrt-sfe-flowoffload-ath79 #54: AUG image does not have the software offload checkmark in firewall settings

Pentaphon, 2019-07-22:
As the title says, there's no checkmark. No way to see if software offload is enabled.

gwlim, 2019-07-22:
It has changed to a button. Isn't that better?
Btw Openwrt Software Offload is broken in Trunk.

Pentaphon, 2019-07-22:
>Btw Openwrt Software Offload is broken in Trunk. 

Is that the reason the checkmark is gone?

>It has changed to a button. Isn't that better?

Don't you mean only SFE has changed to a button? Not software offload?

>Only SFE works and it is faster

I tried it on my Archer C7 v2 and unfortunately, software offloading is faster than SFE on this hardware. I guess I will keep this issue open until software offloading comes back to these builds.

gwlim, 2019-07-22:
I moved the kmod into the tar.gz you can install it from the tar.gz if you want. But the checkbox will not show up.
Just curious, how did you compare?

Pentaphon, 2019-07-22:
In previous versions of your builds when the checkmark was there I tried SFE alone and software offloading alone. Software offloading was without a doubt faster on my Archer C7 v2 so I just kept using software offloading. However, I did a clean install when upgrading to this AUG firmware. I guess I will just have to downgrade to the last version with working software offload.

gwlim, 2019-07-22:
No the question was HOW do you determine which is faster.

Pentaphon, 2019-07-22:
I did numerous speed tests with each of them enabled alone, and none enabled. Software offloading gave me the fastest throughput every single time on the Archer C7 v2.  Not sure why. Perhaps software offloading is the only stable solution for this hardware right now.

gwlim, 2019-07-22:
> I did numerous speed tests with each of them enabled alone, and none enabled. Software offloading alone gave me the fastest throughput every single time on the Archer C7 v2. Not sure why. Perhaps software offloading is the only stable solution for this hardware right now.

It is not so simple. I am not sure you did the test correctly.
I certainly hope you didn't use speedtest as a benchmark.
And SFE actually works with SQM but Flow Offload doesn't.
It is actually a foregone conclusion that SFE is faster
1) It uses prefetch to optimize packet processing
2) It uses aligned to optimize memory accesss
3) I tested using iperf/jperf countless times and SFE is definitely a magnitude faster.

Pentaphon, 2019-07-22:
Did you test it on the Archer C7 v2? Because I honestly get the opposite result with this hardware.

gwlim, 2019-07-22:
> Did you test it on the Archer C7 v2? Because I honestly get the opposite result with this hardware.

I have C7 v4 and WDR4300 both running the MIPS74k CPU at different frequencies.

Pentaphon, 2019-07-22:
I don't use SQM because it slows down everything and it has never fixed bufferbloat for me on my Archer C7 v2. I tried Flow Offload alone and SFE alone, and Flow Offload always gave me a better result in real world performance. When I enable SFE on a clean install of your AUG build, I cannot even get to my 200mbps line speed, while Flow Offload always fulfills my line speed and beyond. I cannot explain why other than online speedtests, real world download performance, and transferring files via the network from device to device.

isbecker, 2019-07-24:
So, does this mean that the process for enabling SFE no longer requires going into the Network->Firewall to click the checkbox?

Meaning, just going into System->Startup and Enabling, then starting SFE (and reboot).

I do see in the syslog:
```
Mon Jul 22 16:29:59 2019 kern.alert kernel: [   27.245457] fast-classifier: starting up
Mon Jul 22 16:29:59 2019 kern.alert kernel: [   27.249627] fast-classifier: registered
```
So, does that meaning that SFE is on and running?

gwlim, 2019-08-03:
You can see the button on the firewall page. FlowOffload kmods are in the tar.gz if you want to use FlowOffload download and install from the zip archive.

closed 2019-08-03 by gwlim