gwlim/openwrt-sfe-flowoffload-ath79 #55: TP-Link WR1043NDv1-32MB-RAM-mips24k-ath9k

Vik-Osa, 2019-07-24:
openwrt-sfe-flowoffload/AUG-2019/openwrt-sfe-flowoffload-mini/TP-Link WR1043NDv1-32MB-RAM-mips24k-ath9k/openwrt-ath79-generic-tplink_tl-wr1043nd-v1-squashfs-factory.bin

Firmware installed. The router has earned. Problem: Does not save settings in LuCI. Such an impression that there is not enough memory. Hangs on trying to save the changed settings. Only after the router itself cleared its memory, closing unnecessary programs, was it possible to partially save the settings.

Vik-Osa, 2019-07-24:
Ready to help test:
openwrt-sfe-flowoffload-mini/TP-Link WR1043NDv1-32MB-RAM-mips24k-ath9k/openwrt-ath79-generic-tplink_tl-wr1043nd-v1-squashfs-factory.bin

gwlim, 2019-08-03:
Issue should be fixed, tested with IPv6 ISP provider.

closed 2019-08-03 by gwlim