gwlim/openwrt-sfe-flowoffload-ath79 #56: WLAN tx power problem

kolomparrudi, 2019-07-26:
Hi,

I have problem setting up tx power level for the 2.4GHz Wifi. I can set the choosen dbm level in Luci, but the it will switch back to auto and the power level remains 15 dBm in TL-WR1043NDv3 and 17 dBm in WDR4300 (iwinfo also reports the same dBm)

In WDR4300 the 5GHz Wifi tx power level seems to be fine, but the luci select list is also switches to auto.

Thanks,
kolomparrudi

kolomparrudi, 2019-07-27:
It seems ART partition regulate based on country (or authority like FCC, ETSI), because with US and CA countries my TL-WR1043NDv3 can transfer with 25 dBm and my WDR4300 can 19 dBm.
I have an custom regulatory.db, but it seems it's not enough.

gwlim, 2019-07-28:
That should be a LuCI issue try with Trunk and raise it up with LuCI developers.

gwlim, 2019-08-12:
https://github.com/openwrt/luci/issues/2626

closed 2019-08-12 by gwlim