gwlim/openwrt-sfe-flowoffload-ath79 #61: Port forwarding broken when SFE enabled on latest AUG build

Patazerty, 2019-08-07:
After updating from the previous AUG build to the current one on my archer C7, I noticed port forwarding was breaking as soon as SFE kicks in (after 4 packets). Using netcat to debug, the connection would establish and a few packets can arrive, but then when SFE picks up the connection, nothing would arrive until one side closes the connection. Reverting to the previous AUG build and port forwarding is working fine again (but no ipv6 accelerated with SFE).

gwlim, 2019-08-07:
I am able to setup a local port forwarding setup both TCP and UDP using JPerf, you have to give me more details for this.

gwlim, 2019-08-08:
I tested port forward on my WAN link, no problem reaching the internal ip & port as far as I can tell.
I will close this ticket until more information is given

closed 2019-08-08 by gwlim

Xmister, 2019-08-09:
I have the same issue (Archer C7 v4), although my case a little bit special:
I have another router behind my openwrt router that is also doing NAT.
Reaching internet behind the second router works fine, but if I forward a port from the sfe router to the second router it's like the MSS Clamping rule wouldn't work, as in small packets get through the connection but bigger ones are getting dropped and retransmitted. (I'm using PPoE)

Patazerty, 2019-08-09:
I can only access the router remotely for now, will do more testing in a few days

gwlim, 2019-08-11:
> @gwlim
> I have the same issue (Archer C7 v4), although my case a little bit special:
> I have another router behind my openwrt router that is also doing NAT.
> Reaching internet behind the second router works fine, but if I forward a port from the sfe router to the second router it's like the MSS Clamping rule wouldn't work, as in small packets get through the connection but bigger ones are getting dropped and retransmitted. (I'm using PPoE)

Try setting MTU = 1492?

Xmister, 2019-08-11:
> Try setting MTU = 1492?

I've tried that before, but didn't help. I'm not sure if the MSS is the actual issue, it just has very similar symptoms: Small packages go through, big ones don't. Like HTTP works, but HTTPS doesn't, because of the larger SSL packets.

gwlim, 2019-08-12:
> I've tried that before, but didn't help. I'm not sure if the MSS is the actual issue, it just has very similar symptoms: Small packages go through, big ones don't. Like HTTP works, but HTTPS [...]

Nope SSL Packets sizes are the same. 
I just tested port forward external port 6001 to second AP on https port 443.
Works fine.

Xmister, 2019-08-12:
Well, SSL packets usually have the DF bit set, so they tend to arrive in bigger chunks.
Anyway, what kind of log/test would you need to help in debugging?

gwlim, 2019-08-12:
I just need a reliable way to reproduce this.
A simple test case that demonstrate this

Xmister, 2019-08-12:
This is the route of the connection:

```
Incoming SSH      Port Forward to 10.0.1.2     Port Forward to 10.0.3.10
     PPPOE WAN ----------> SFE Router ------------------> Other router ------------------> Destination
                       <wan ip> / 10.0.1.1            10.0.1.2 / 10.0.3.3                    10.0.3.10
```
The ports are always 22.

gwlim, 2019-08-13:
Update: 
I am able to reproduce the issue, and the issue has to do with which route SFE chooses first.
The fix seems to work, right now I am running it on my main router to see if there is any issues.
Will probably be in the next release if there are no issues.

Xmister, 2019-08-13:
Great news, thank you!

0sy, 2019-08-14:
Hi gwlim - thank you so much for your work. Same issue here, on Tp-Link WR-1043ND v2.

Quick question here to you and the others experiencing it - would disabling SFE allow port forwarding to function as intended? CC @Patazerty @Xmister @gwlim

Xmister, 2019-08-14:
@0sy Yes, disabling SFE solves the issue for me.

Zsolteeca, 2019-08-15:
> 
> 
> Update:
> I am able to reproduce the issue, and the issue has to do with which route SFE chooses first.
> The fix seems to work, right now I am running it on my main router to see if there is any issues.
> Will probably be in the next release if there are no issues.

When will be the next release?

Zsolteeca, 2019-08-18:
I am looking forward to the new version of your firmware

Xmister, 2019-08-28:
Works fine after the SEP update, thank you!