gwlim/openwrt-sfe-flowoffload-ath79 #64: Support an new router.

Axius, 2019-08-13:
Can you please add support for TP-Link Archer C6v2.
That would be greatly appreciated.

Thanks,

kolomparrudi, 2019-08-26:
@gwlim: it would be fine if you can share (or update mips74k-ar71xx-lede-patch repo) your current script and sources for patching to able to add new router by the community. I have a new Xiaomi router 3G and I have issues with the latest openwrt flow offload feature (remote desktop lagging, disconnectiong) and I'd like to try sfe there.

Thanks,

gwlim, 2019-08-27:
> @gwlim: it would be fine if you can share (or update mips74k-ar71xx-lede-patch repo) your current script and sources for patching to able to add new router by the community. I have a new Xiaomi [...]

I am working on mips1004kc as I have a Newifi-D2

> 
> 
> Can you please add support for TP-Link Archer C6v2.
> That would be greatly appreciated.
> 
> Thanks,

Added in SEP Image

closed 2019-08-27 by gwlim