gwlim/openwrt-sfe-flowoffload-ath79 #67: WDR4900 Aug build package manager fails

Dean-Bayliss, 2019-08-25:
Attempting to load packages gives:

_Downloading http://downloads.openwrt.org/snapshots/targets/mpc85xx/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/mpc85xx/generic/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
- all fail.  /usr/sbin/opkg-key: line 22: usign: not found_

Re-loading March build which I was on and package manager works fine.  I suspect this is specific to the WDR4900 as some changes have been made to manage the image size.

Dean-Bayliss, 2019-08-25:
Just downloaded the snapshot 
**Firmware Version	OpenWrt SNAPSHOT r10863-e1dcfe02b2 / LuCI Master (git-19.231.47191-2aab858)
Kernel Version	4.19.66**
==> An attempt at updating the packages does operate correctly.

Dean-Bayliss, 2019-08-26:
Thanks - opkg.conf >option check_signature removed - fixed

closed 2019-08-26 by Dean-Bayliss

gwlim, 2019-08-26:
Try to remove package signing config.
I have remove package signing in order to conserve space since it is seldom used, people don't install packages on a daily basis.