gwlim/openwrt-sfe-flowoffload-ath79 #68: New version?

Zsolteeca, 2019-08-25:
I am looking forward to see the new version of Fastpath Lede.

Thank you.

Patazerty, 2019-08-26:
Just a friendly reminder: this is not a commercial project, but just a project maintained by a single person that is probably doing it on his/her free time. You can't expect a commercial-grade level of support for a community project like this. You've already posted multiple comments asking for an ETA for a new version, you don't have to open a new issue for that. 
You can downgrade, build your own software, or maybe simply don't buy consumer-grade hardware to install a unofficial build of OpenWRT made by someone you don't know if you "can't live" without one feature that is essentially a hack to mitigate the relative low performance of consumer routers.

Zsolteeca, 2019-08-27:
Port forwarding is a very important thing everyone who has to use use some services on their computers.. 
The latest version has got problems with it and this needs to be "repaired" 
I always say thank to the author because of his work..

closed 2019-08-27 by Zsolteeca