gwlim/openwrt-sfe-flowoffload-ath79 #70: Kernel 4.19.65 release date?

oriceon, 2019-08-29:
Hi there. Any ideea when sfe* builds will be released with 4.19.65 support ?

Thanks.

gwlim, 2019-08-29:
Anything significant about that version?

oriceon, 2019-08-29:
Hi, i`m trying to manual install those that they need .65+ kernel :(

```
kmod-loop
kmod-crypto-sha256
kmod-crypto-sha512
kmod-ledtrig-heartbeat
kmod-ledtrig-netdev
kmod-ipt-compat-xtables kmod-sit
kmod-usb-serial
kmod-usb-serial-option
kmod-usb-storage
kmod-usb-storage-extras
kmod-usb-uhci
kmod-usb3
6in4
```

thomasrc, 2019-08-29:
You can install them from the tarball archive (kmod-package.tar.gz).

Just copy the necessary modules to the router and use `opkg install` command:

`opkg install /tmp/kmod-whatever.ipk`

MadRock88, 2019-09-09:
same trouble on sept test image (wdr4300) with kmod-usb-storage package from tarball arch.
archive has cc50 revision, but kernel in firmware has another revision.
and no one say, can i force install this ipk or not.
becouse revision its good, but maybe modules which are needed in cc50rev ipk not changed on latest firmware, and links to them to, and this mistmatch on kernels only paper error.
but i dont want try it on work device T_T

closed 2019-12-19 by gwlim