gwlim/openwrt-sfe-flowoffload-ath79 #71: Slow download speed using openwrt-sfe.

Axius, 2019-08-29:
Hello,
I have a problem with the download speed using  TP-Link Archer C6 v2 (EU).
I have gigabit connection form my ISP.

The speeds I get now -> 130 Mbps Download
                                   -> 415 Mbps Upload.

Can you help me fix this problem.

Xmister, 2019-08-29:
Do you have SFE enabled in Network/Firewall?
[image]

Axius, 2019-08-29:
I've enabled SFE flow offloading.

Axius, 2019-08-29:
Do I have any options this fix this problem?

gwlim, 2019-08-30:
Reboot after enabling SFE
If you are using speedtest try changing servers.
Test using direct connection from computer to modem

gwlim, 2019-09-02:
I will be closing this if there is no response.

closed 2019-09-02 by gwlim