gwlim/openwrt-sfe-flowoffload-ath79 #72: Possible memory leak on Archer C7v4

Xmister, 2019-08-30:
Hi!
Something eats up the memory on my router over time. I haven't installed anything other than the base system, and from `htop` I can't see any process with excess usage, so it must be somthing in the kernel.
When I reboot it, I have ~75MB free.
After each day, it goes down about ~40MB.
At the 2nd day, I have to reboot the router because even luci crashes with out of memory errors.
```bash
root@OpenWrt:~# uptime 
 09:20:47 up 16:44,  load average: 0.25, 0.19, 0.11
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         124260       76340       33852        1304       14068       16712
Swap:             0           0           0
root@OpenWrt:~# vmstat 
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 0  0      0  37180   4192  11832    0    0     0     0  425  246  3  8 89  0  0
root@OpenWrt:~# cat /proc/meminfo 
MemTotal:         124260 kB
MemFree:           36708 kB
MemAvailable:      19532 kB
Buffers:            4192 kB
Cached:             9876 kB
SwapCached:            0 kB
Active:            10712 kB
Inactive:           5272 kB
Active(anon):       3184 kB
Inactive(anon):       36 kB
Active(file):       7528 kB
Inactive(file):     5236 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          1928 kB
Mapped:             2936 kB
Shmem:              1304 kB
Slab:              40912 kB
SReclaimable:       2032 kB
SUnreclaim:        38880 kB
KernelStack:         360 kB
PageTables:          316 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       62128 kB
Committed_AS:       6316 kB
VmallocTotal:    1048372 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:               32 kB
```

gwlim, 2019-08-30:
What process do you have running?

Xmister, 2019-08-30:
Updated my command with `ps`

gwlim, 2019-08-30:
I have what you have except pppd and crond

Xmister, 2019-08-30:
Stopping crond doesn't really make a change:
```bash
root@OpenWrt:~# /etc/init.d/cron stop
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         124260       74616       35564        1304       14080       18400
Swap:             0           0           0
root@OpenWrt:~# ps | grep cron
29827 root      1432 S    grep cron
root@OpenWrt:~# 
```
And from `htop` I can see that `pppd` doesn't use much memory either:
```bash
  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command
 1600 root       20   0  1288   856   740 S  0.0  0.7  0:00.40 /usr/sbin/pppd nodetach ipparam wan ifname pppoe-wan nodefaultroute usepeerdns
```

gwlim, 2019-08-30:
when memory leaks the os doesn't see it.
If it could see it then the memory would be able to be reclaimed.
Does sysctl -w vm.drop_caches=2 reclaim anything?

Xmister, 2019-08-30:
Doesn't really help:
```bash
root@OpenWrt:~# sysctl -w vm.drop_caches=2
vm.drop_caches = 2
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75268       40192        1304        8800       20084
Swap:             0           0           0
root@OpenWrt:~#
```
I didn't have this issue in vanilla OpenWrt, so I doubt that any standard package would cause this.

gwlim, 2019-08-30:
Try stopping the additional services to see if it still leaks.
This suggestion is not a fix but fault isolation.

Xmister, 2019-08-30:
After disabling shortcut-fe in luci, and
```bash
root@OpenWrt:/# rmmod shortcut_fe
root@OpenWrt:/# rmmod shortcut_fe_ipv6
root@OpenWrt:/# /etc/init.d/cron start
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75036       40232        1304        8992       20220
Swap:             0           0           0
```
One hour later:
```bash
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       74904       40360        1304        8996       20352
Swap:             0           0           0
```
The memory leak stops.

gwlim, 2019-08-30:
How many connections do you have in  cat /sys/fast_classifier/debug_info

Xmister, 2019-08-30:
Here is the memory usage and debug_info combined over the course of 30 min:
```bash
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       74512       35884        1304       13864       18380
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=410 offload=0 offload_no_match=0 offloaded=483 done=183 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       74472       35924        1304       13864       18420
Swap:             0           0           0
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       76184       34212        1304       13864       16708
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=473 offload=0 offload_no_match=0 offloaded=1173 done=840 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75476       34920        1304       13864       17416
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=439 offload=0 offload_no_match=0 offloaded=1247 done=941 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75372       35024        1304       13864       17520
Swap:             0           0           0
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75460       34936        1304       13864       17432
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=439 offload=0 offload_no_match=0 offloaded=1721 done=1401 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=469 offload=0 offload_no_match=0 offloaded=1766 done=1446 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75460       34936        1304       13864       17432
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=423 offload=0 offload_no_match=0 offloaded=1965 done=1670 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75656       34740        1304       13864       17236
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=438 offload=0 offload_no_match=0 offloaded=2148 done=1839 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75708       34688        1304       13864       17184
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=449 offload=0 offload_no_match=0 offloaded=2500 done=2169 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       75848       34548        1304       13864       17044
Swap:             0           0           0
root@OpenWrt:/# head -1 /sys/fast_classifier/debug_info && cat /sys/fast_classifier/debug_info | wc -l
size=402 offload=0 offload_no_match=0 offloaded=3535 done=3246 offloaded_fail=0 done_fail=0 
39
root@OpenWrt:/# free
              total        used        free      shared  buff/cache   available
Mem:         124260       76244       34152        1304       13864       16648
Swap:             0           0           0
```

gwlim, 2019-08-30:
Please try the test fix I just uploaded

Xmister, 2019-08-30:
Running that now. Will report back in a few hours.

gwlim, 2019-08-30:
> 
> 
> Running that now. Will report back in a few hours.

Sry do not try that

Xmister, 2019-08-30:
Running that now. I'll report back tomorrow.

Xmister, 2019-08-31:
After 14 hours of uptime it looks promising:
```bash
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         124260       39960       65724         600       18576       51428
Swap:             0           0           0

```
But I would like to keep running it a day or more to be sure.

gwlim, 2019-09-01:
> 
> 
> After 14 hours of uptime it looks promising:
> 
> ```shell
> root@OpenWrt:~# free
>               total        used        free      shared  buff/cache   available
> Mem:         124260       39960       65724         600       18576       51428
> Swap:             0           0           0
> ```
> 
> But I would like to keep running it a day or more to be sure.

Close this ticket when you think that it is okay and I will push the changes to all images.

Xmister, 2019-09-01:
After another day, the values are basically the same, so it seems the issue is indeed fixed:
```bash
root@OpenWrt:~# free
              total        used        free      shared  buff/cache   available
Mem:         124260       39708       65976         600       18576       51684
Swap:             0           0           0
```
Thank you, @gwlim!

closed 2019-09-01 by Xmister