gwlim/openwrt-sfe-flowoffload-ath79 #74: Wdr4300 after enabling SFE no internet

vic1234, 2019-09-01:
If I enable SFE and reboot
All OK for few minutes after router is reboots
But then the internet disappear 
The connection to the router from the pc is working
But if I reload the Wan port in luci I not recieve an IP from the modem
When I leave the SFE disabled all is working good
This problem appears after installing the latest build

gwlim, 2019-09-02:
what is your wan connection type?
any logs?

vic1234, 2019-09-02:
What do you mean by Wan connection type?
DHCP with cable modem
Didn't tried ping but can't access the modem web interface 
And no Wan IP 

If I not mistaken the modem blocks connections if the Mac that connected to it is changed few times

gwlim, 2019-09-02:
You can try
1)Power Off Cable Modem with Router for 5 mins then power on again.
2)Contact your ISP for them to release the MAC manually.

vic1234, 2019-09-02:
The power off is working for few minutes and than again the Wan connection dissappears

The Mac block goes off when I reboot my modem

When the SFE is disabled all working good
What log do you need kernel or system?

gwlim, 2019-09-02:
Have you tried reset the router back to factory and re configuring the interface?

vic1234, 2019-09-02:
Didn't tried reset the router to factory to avoid reconfiguration of everything
Didn't changee the MTU 1500

https://i.ibb.co/4Ttk8dm/Screenshot-20190902-151911-Chrome-Beta.jpg

I will try to add log later

gwlim, 2019-09-02:
You can try backup your current config do a full reset and test the wan interface only

vic1234, 2019-09-02:
here is a ping test from the openwrt interface after enable SFE
to the modem (the internet stiil worked)
PING 192.168.100.1 (192.168.100.1): 56 data bytes
64 bytes from 192.168.100.1: seq=0 ttl=64 time=0.722 ms
64 bytes from 192.168.100.1: seq=1 ttl=64 time=0.688 ms
64 bytes from 192.168.100.1: seq=2 ttl=64 time=0.617 ms
64 bytes from 192.168.100.1: seq=3 ttl=64 time=0.625 ms
64 bytes from 192.168.100.1: seq=4 ttl=64 time=0.593 ms

--- 192.168.100.1 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 0.593/0.649/0.722 ms

and ping to google.com
PING google.com (172.217.21.206): 56 data bytes
64 bytes from 172.217.21.206: seq=0 ttl=56 time=2808.353 ms
64 bytes from 172.217.21.206: seq=1 ttl=56 time=2528.023 ms
64 bytes from 172.217.21.206: seq=2 ttl=56 time=1705.331 ms
64 bytes from 172.217.21.206: seq=3 ttl=56 time=820.589 ms
64 bytes from 172.217.21.206: seq=4 ttl=56 time=949.012 ms

--- google.com ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 820.589/1762.261/2808.353 ms

here is a ping after the sfe was disabled

PING google.com (216.58.208.46): 56 data bytes
64 bytes from 216.58.208.46: seq=0 ttl=56 time=66.017 ms
64 bytes from 216.58.208.46: seq=1 ttl=56 time=64.164 ms
64 bytes from 216.58.208.46: seq=2 ttl=56 time=63.991 ms
64 bytes from 216.58.208.46: seq=3 ttl=56 time=63.567 ms
64 bytes from 216.58.208.46: seq=4 ttl=56 time=63.767 ms

--- google.com ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 63.567/64.301/66.017 ms

PING 172.217.21.206 (172.217.21.206): 56 data bytes
64 bytes from 172.217.21.206: seq=0 ttl=56 time=67.098 ms
64 bytes from 172.217.21.206: seq=1 ttl=56 time=66.298 ms
64 bytes from 172.217.21.206: seq=2 ttl=56 time=69.371 ms
64 bytes from 172.217.21.206: seq=3 ttl=56 time=66.681 ms
64 bytes from 172.217.21.206: seq=4 ttl=56 time=68.257 ms

--- 172.217.21.206 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 66.298/67.541/69.371 ms

gwlim, 2019-09-02:
cat /sys/fast_classifier/debug_info?

vic1234, 2019-09-02:
I will do it all to night when no one will use the internet

vic1234, 2019-09-02:
here is a log after enabling SFE
internet stopped working after few minutes
no ping to modem
 00:38:55 2019 daemon.notice miniupnpd[1872]: DNS rebinding attack suspected (Host: :*::5000)
Tue Sep  3 00:46:00 2019 daemon.err uhttpd[1220]: luci: accepted login on / for root from 192.168.1.162
Tue Sep  3 00:49:03 2019 authpriv.info dropbear[27973]: Child connection from 192.168.1.162:54544
Tue Sep  3 00:49:12 2019 kern.alert kernel: [31998.511773] fast-classifier: starting up
Tue Sep  3 00:49:12 2019 kern.alert kernel: [31998.516034] fast-classifier: registered
Tue Sep  3 00:49:31 2019 authpriv.notice dropbear[27973]: Password auth succeeded for 'root' from 192.168.1.162:54544
Tue Sep  3 00:49:46 2019 authpriv.info dropbear[28438]: Child connection from 192.168.1.162:54578
Tue Sep  3 00:49:48 2019 authpriv.notice dropbear[28438]: Password auth succeeded for 'root' from 192.168.1.162:54578
Tue Sep  3 00:52:48 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: authenticated
Tue Sep  3 00:52:48 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: associated (aid 3)
Tue Sep  3 00:52:48 2019 daemon.info hostapd: wlan0: STA :*: RADIUS: starting accounting session 72F8DF0F55973AE2
Tue Sep  3 00:52:48 2019 daemon.info hostapd: wlan0: STA :*: WPA: pairwise key handshake completed (RSN)
Tue Sep  3 00:52:48 2019 daemon.info dnsmasq-dhcp[2123]: DHCPDISCOVER(br-lan) :*:
Tue Sep  3 00:52:48 2019 daemon.info dnsmasq-dhcp[2123]: DHCPOFFER(br-lan) 192.168.1.226 :*:
Tue Sep  3 00:52:48 2019 daemon.info dnsmasq-dhcp[2123]: DHCPREQUEST(br-lan) 192.168.1.226 :*:
Tue Sep  3 00:52:48 2019 daemon.info dnsmasq-dhcp[2123]: DHCPACK(br-lan) 192.168.1.226 :*: Redmi4A-Redmi
Tue Sep  3 00:57:01 2019 authpriv.info dropbear[28438]: Exit (root): Exited normally
Tue Sep  3 00:57:11 2019 authpriv.info dropbear[27973]: Exit (root): Exited normally
Tue Sep  3 01:08:59 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: authenticated
Tue Sep  3 01:08:59 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: associated (aid 5)
Tue Sep  3 01:08:59 2019 daemon.info hostapd: wlan0: STA :*: RADIUS: starting accounting session 49A1A70761CF9DA0
Tue Sep  3 01:08:59 2019 daemon.info hostapd: wlan0: STA :*: WPA: pairwise key handshake completed (RSN)
Tue Sep  3 01:08:59 2019 daemon.info dnsmasq-dhcp[2123]: DHCPDISCOVER(br-lan) :*:
Tue Sep  3 01:08:59 2019 daemon.info dnsmasq-dhcp[2123]: DHCPOFFER(br-lan) 192.168.1.222 :*:
Tue Sep  3 01:08:59 2019 daemon.info dnsmasq-dhcp[2123]: DHCPREQUEST(br-lan) 192.168.1.222 :*:
Tue Sep  3 01:08:59 2019 daemon.info dnsmasq-dhcp[2123]: DHCPACK(br-lan) 192.168.1.222 :*: ESP_5E70B1
Tue Sep  3 01:09:58 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: authenticated
Tue Sep  3 01:09:58 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: associated (aid 5)
Tue Sep  3 01:10:00 2019 daemon.info hostapd: wlan0: STA :*: RADIUS: starting accounting session 49A1A70761CF9DA0
Tue Sep  3 01:10:00 2019 daemon.info hostapd: wlan0: STA :*: WPA: pairwise key handshake completed (RSN)
Tue Sep  3 01:10:00 2019 daemon.info dnsmasq-dhcp[2123]: DHCPDISCOVER(br-lan) :*:
Tue Sep  3 01:10:00 2019 daemon.info dnsmasq-dhcp[2123]: DHCPOFFER(br-lan) 192.168.1.222 :*:
Tue Sep  3 01:10:00 2019 daemon.info dnsmasq-dhcp[2123]: DHCPREQUEST(br-lan) 192.168.1.222 :*:
Tue Sep  3 01:10:00 2019 daemon.info dnsmasq-dhcp[2123]: DHCPACK(br-lan) 192.168.1.222 :*: ESP_5E70B1
Tue Sep  3 01:11:01 2019 authpriv.info dropbear[32072]: Child connection from 192.168.1.162:55489
Tue Sep  3 01:11:04 2019 authpriv.warn dropbear[32072]: Bad password attempt for 'root' from 192.168.1.162:55489
Tue Sep  3 01:11:06 2019 authpriv.notice dropbear[32072]: Password auth succeeded for 'root' from 192.168.1.162:55489
Tue Sep  3 01:11:08 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: disassociated
Tue Sep  3 01:11:09 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Sep  3 01:11:10 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: authenticated
Tue Sep  3 01:11:10 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: associated (aid 5)
Tue Sep  3 01:11:10 2019 daemon.info hostapd: wlan0: STA :*: RADIUS: starting accounting session 1E533445CB832FDF
Tue Sep  3 01:11:10 2019 daemon.info hostapd: wlan0: STA :*: WPA: pairwise key handshake completed (RSN)
Tue Sep  3 01:11:10 2019 daemon.info dnsmasq-dhcp[2123]: DHCPDISCOVER(br-lan) :*:
Tue Sep  3 01:11:10 2019 daemon.info dnsmasq-dhcp[2123]: DHCPOFFER(br-lan) 192.168.1.222 :*:
Tue Sep  3 01:11:10 2019 daemon.info dnsmasq-dhcp[2123]: DHCPREQUEST(br-lan) 192.168.1.222 :*:
Tue Sep  3 01:11:10 2019 daemon.info dnsmasq-dhcp[2123]: DHCPACK(br-lan) 192.168.1.222 :*: ESP_5E70B1
Tue Sep  3 01:12:20 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: disassociated
Tue Sep  3 01:12:21 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Sep  3 01:12:22 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: authenticated
Tue Sep  3 01:12:22 2019 daemon.info hostapd: wlan0: STA :*: IEEE 802.11: associated (aid 5)
Tue Sep  3 01:12:23 2019 daemon.info hostapd: wlan0: STA :*: RADIUS: starting accounting session EA58D6F327EB7F11
Tue Sep  3 01:12:23 2019 daemon.info hostapd: wlan0: STA :*: WPA: pairwise key handshake completed (RSN)
Tue Sep  3 01:12:23 2019 daemon.info dnsmasq-dhcp[2123]: DHCPDISCOVER(br-lan) :*:
Tue Sep  3 01:12:23 2019 daemon.info dnsmasq-dhcp[2123]: DHCPOFFER(br-lan) 192.168.1.222 :*:
[... cut ...]
o=1, p=6 [*0.*]:81.200.8.228:18406 *:[*0.*] m=00000000 h=4
o=0, p=6 [*0.*]:192.168.1.101:44530 *[*0.*] m=00000000 h=3
o=1, p=6 [*0.*]:192.168.1.119:38877 :*:[*0.*] m=00000000 h=4
o=1, p=6 [*0.*]::[*0.*] m=00000000 h=4 
o=1, p=6 [*0.*]::[*0.*] m=00000000 h=10 
o=1, p=6 [*0.*]:192.168.1.142:59364 :[*0.*] m=00000000 h=4
o=1, p=6 [*0.*]:192.168.1.248:49115 7:*:[*0.*] m=00000000 h=4
o=1, p=6 [*0.*]:*:[*0.*] m=00000000 h=4 
o=1, p=6 [*0.*]:192.168.1.142:49780 
 /sys/kernel/debug/mips/unaligned_instructions
is empty

vic1234, 2019-09-02:
reboot to the router doesnt bring back the internet
only when i reboot the router and modem the internet comes back for few minutes

vic1234, 2019-09-02:
this appears after i lost the internet and tried to restart the wan interface through luci
Tue Sep 3 01:16:14 2019 daemon.notice netifd: Interface 'wan' is now down

gwlim, 2019-09-03:
try disable miniipnpd?
also go to backup configuration, create backup
reset to factory then try connect to wan bare metal configuration

gwlim, 2019-09-03:
I think I might know what is happening

vic1234, 2019-09-03:
The internet works fine when the SFE is disabled
So it can't be a cable problem or setting no?

gwlim, 2019-09-03:
> 
> 
> The internet works fine when the SFE is disabled
> So it can't be a cable problem or setting no?
> 
> Previous builds worked fine

I can't reproduce the problem what special config do you use?

vic1234, 2019-09-03:
Last build that was fine is Aug build
No special CONFIG didn't change anything in Wan settings Just configed as DHCP

In the log there is no change after I enable SFE 
Only this line appears
Tue Sep 3 00:49:12 2019 kern.alert kernel: [31998.511773] fast-classifier: starting up
Tue Sep 3 00:49:12 2019 kern.alert kernel: [31998.516034] fast-classifier: registered

And nothing else only wifi devices connection log

Willem1981, 2019-09-04:
Same issue on a TP-Link Archer C7 v2.
After enabling SFE it will not load any internet pages.
When disabled the loading directly works again.

gwlim, 2019-09-06:
Add test image

Pentaphon, 2019-09-06:
> After enabling SFE it will not load any internet pages.

I've never had this problem on my TP-Link Archer C7 v2 but I will maintain that flow offloading works better than SFE on this hardware.

gwlim, 2019-09-06:
> 
> 
> > After enabling SFE it will not load any internet pages.
> 
> I've never had this problem on my TP-Link Archer C7 v2 but I will maintain that flow offloading works better than SFE on this hardware.

Sure you are free to use flow offloading then

Axius, 2019-09-06:
How can one enable flow offloading on SFE?

gwlim, 2019-09-06:
> 
> How can one enable flow offloading on SFE?
> […](#)

Download from Openwrt.org

Axius, 2019-09-06:
What packages do I need for flow offloading to work on SFE?

kolomparrudi, 2019-09-06:
You can't use Flow offload with SFE builds, because it based on Openwrt Trunk with kernel 4.19 and Flow offload is broken with kernel 4.19, you can check the current status of the kernel ticket here: https://bugzilla.kernel.org/show_bug.cgi?id=204507

Willem1981, 2019-09-06:
Have installed the test. It works but after installing 6RD it fails. Then it only works with SFE disabled. Prehaps it something with IPv6 over 6RD

gwlim, 2019-09-07:
Can you guys be clear what is the config being use?
I keep hearing new things "failing" which was never being mentioned before.
I can only test if I know

vic1234, 2019-09-07:
what config do you need to know?
here is the network config
`
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd49:69f6:c2ee::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'
	option peerdns '0'
	option dns '8.8.4.4 8.8.8.8'

config interface 'wan6'
	option ifname 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '2 3 4 5 0t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '1 0t'

`
The router is connected to docsis 3 cable modem router that configured in bridge mode.
that meaning it working as a modem

gwlim, 2019-09-07:
Have you tried the test image?

vic1234, 2019-09-07:
not yet

Willem1981, 2019-09-08:
> Can you guys be clear what is the config being use?
> I keep hearing new things "failing" which was never being mentioned before.
> I can only test if I know funny part is whatever issues some people have, they told me it is standard config but I am unable to reproduce it.

Please tell me what info you need. After installing 6RD for IPV6 from my provider Then when the SFE enabled i cannot go to new webpages. The pages that are already opened in the browser work correctly and keep on working.

gwlim, 2019-09-15:
Release reworked test image please test and feedback

vic1234, 2019-09-16:
installed the reworked version
all seems to work for now

thank you

gwlim, 2019-09-16:
@Willem1981 
Can you test for 6rd?

Willem1981, 2019-09-16:
I have tested the latest test build.
Its a clean install.
Without 6RD it works fine.
After installing 6RD the browser give a site unavailable please check connection, proxy, firewall. ERR_CONNECTION_RESET.

I also see that i can ping the webaddress using ip6 but when i force to ping with ip4 it cannot and i get a request timed out.
I have tested this on the computer and with putty on the router. Both the same result.

ip4 only websites still work after installing 6RD.

gwlim, 2019-09-16:
> I have tested the latest test build.
> Its a clean install.
> Without 6RD it works fine.
> After installing 6RD the browser give a site unavailable please check connection, proxy, firewall. ERR_CONNECTION_RESET.
> 
> I also see that i can ping the webaddress using ip6 but when i force to ping with ip4 it cannot and i get a request timed out.
> I have tested this on the computer and with putty on the router. Both the same result.
> 
> ip4 only websites still work after installing 6RD.
> 
> I came from the March version. I have installed the Aug version but that one does also not work with 6RD enabled.
Do you have any ipv6 dns servers?
Is the ipv6 addresses resolving correctly?
Do you have any free 6rd service to test?

Willem1981, 2019-09-16:
I use DNS servers from 1.1.1.1 and for ipv6 2606:4700:4700::1111

IPv6 and IPv6 both work without SFE.
As soon as SFE is on i can ping both ipv4 and ipv6 addresses but the browser is showing no connection.

Using discord and teamviewer you may see the config if you like.

gwlim, 2019-09-16:
ok when  you
ping ipv6.google.com -> 2a00:1450:400a:803::200e
So in
cat /sys/fast-classifier/debug_info | grep '2a00'
do you see the connection?

Willem1981, 2019-09-16:
> ok when you
> ping ipv6.google.com -> 2a00:1450:400a:803::200e
> So in
> cat /sys/fast-classifier/debug_info | grep '2a00'
> do you see the connection?

When i enable SFE and i ping to ipv6.google.com i can ping from both laptop and router.
When running the command "cat /sys/fast-classifier/debug_info | grep '2a00'" i get "cat: can't open '/sys/fast-classifier/debug_info': No such file or directory"
There is nog fast-classifier directory on the router in sys

gwlim, 2019-09-16:
> 
> 
> > ok when you
> > ping ipv6.google.com -> 2a00:1450:400a:803::200e
> > So in
> > cat /sys/fast-classifier/debug_info | grep '2a00'
> > do you see the connection?
> 
> When i enable SFE and i ping to ipv6.google.com i can ping from both laptop and router.
> When running the command "cat /sys/fast-classifier/debug_info | grep '2a00'" i get "cat: can't open '/sys/fast-classifier/debug_info': No such file or directory"
> There is no fast-classifier directory on the router in sys

You have to start fast-classifier to obtain debugging information
/etc/init.d/shortcut-fe start
Also what router are you using?

Willem1981, 2019-09-16:
TP-Link Archer C7 v2

/etc/init.d/shortcut-fe start > works
cat /sys/fast-classifier/debug_info | grep '2a00' > Does not work ('/sys/fast-classifier/debug_info': No such file or directory")

Ping to ipv4.google.com > works
Ping to ipv6.google.com > works
Browse to ipv4.google.com > works
Browse to ipv6.google.com > does not work

With SFE off all works correctly but with slower speed.

gwlim, 2019-09-16:
> TP-Link Archer C7 v2
>
> /etc/init.d/shortcut-fe start > works
> cat /sys/fast-classifier/debug_info | grep '2a00' > Does not work ('/sys/fast-classifier/debug_info': No such file or directory")
>
> Ping to [...]

cat /sys/fast_classifier/debug_info

gwlim, 2019-09-16:
I think your issue has to do with protocol 41 tunneling, normal ipv6 goes through IPv6 route but protocol 41 requires tunneling to a gateway
I think I need to setup a 6in4 to troubleshoot this weekend

Willem1981, 2019-09-16:
> I think your issue has to do with protocol 41 tunneling, normal ipv6 goes through IPv6 route but protocol 41 requires tunneling to a gateway
> I think I need to setup a 6in4 to troubleshoot this weekend

root@OpenWrt:~# ping ipv6.google.com
PING ipv6.google.com (2a00:1450:400e:80d::200e): 56 data bytes
64 bytes from 2a00:1450:400e:80d::200e: seq=0 ttl=57 time=4.813 ms
64 bytes from 2a00:1450:400e:80d::200e: seq=1 ttl=57 time=4.780 ms
--- ipv6.google.com ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 4.780/4.796/4.813 ms
root@OpenWrt:~# cat /sys/fast_classifier/debug_info | grep '2a00'
o=0, p=6 [a0:a8:cd:b4:bf:e0]:2002:54f5:094c:0000:4447:6f86:ab65:55b6:53002 2a00:1450:400e:080d:0000:0000:0000:200e:80:[00:00:00:00:00:00] m=00000000 h=3
o=1, p=6 [a0:a8:cd:b4:bf:e0]:2002:54f5:094c:0000:4447:6f86:ab65:55b6:53003 2a00:1450:400e:080d:0000:0000:0000:200e:80:[00:00:00:00:00:00] m=00000000 h=4

SFE is on.

gwlim, 2019-09-16:
> SFE is on.

Yep if you didn't change the mac address it is [00:00:00:00:00:00] meaning it didn't find the correct route it needs to go though.
I need to investigate 6in4 further

Willem1981, 2019-09-16:
> > SFE is on.
> 
> Yep if you didn't change the mac address it is [00:00:00:00:00:00] meaning it didn't find the correct route it needs to go though.
> I need to investigate 6in4 further

Ok, if you need more info or test just let me know.

vic1234, 2019-09-17:
after a wile the pings are high again with SFE on
disabled it for now

same 00 MAC as in Willem1981 log when using the comand 
cat /sys/fast_classifier/debug_info | grep '2a00'

***
not related
when i go to http://192.168.1.1/cgi-bin/luci/admin/status/realtime/bandwidth

i get this
`Failed to execute template dispatcher target for entry '/admin/status/realtime/bandwidth'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/bandwidth'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/admin_status/bandwid..."]:2: attempt to index a boolean value
stack traceback:
	[string "/usr/lib/lua/luci/view/admin_status/bandwid..."]:2: in main chunk
stack traceback:
	[C]: in function 'n'
	/usr/lib/lua/luci/template.lua:55: in function </usr/lib/lua/luci/template.lua:47>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:680: in function </usr/lib/lua/luci/dispatcher.lua:679>`

gwlim, 2019-09-18:
> 
> 
> after a wile the pings are high again with SFE on
> disabled it for now
> 
> same 00 MAC as in Willem1981 log when using the comand
> cat /sys/fast_classifier/debug_info | grep '2a00'
> 
> not related
> when i go to http://192.168.1.1/cgi-bin/luci/admin/status/realtime/bandwidth
> 
> i get this
> `Failed to execute template dispatcher target for entry '/admin/status/realtime/bandwidth'. The called action terminated with an exception: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/bandwidth'. A runtime error occurred: [string "/usr/lib/lua/luci/view/admin_status/bandwid..."]:2: attempt to index a boolean value stack traceback: [string "/usr/lib/lua/luci/view/admin_status/bandwid..."]:2: in main chunk stack traceback: [C]: in function 'n' /usr/lib/lua/luci/template.lua:55: in function </usr/lib/lua/luci/template.lua:47> (tail call): ? /usr/lib/lua/luci/dispatcher.lua:680: in function </usr/lib/lua/luci/dispatcher.lua:679>`
> 
> and network SQM
> http://192.168.1.1/cgi-bin/luci/admin/network/sqm
> `Failed to execute cbi dispatcher target for entry '/admin/network/sqm'. The called action terminated with an exception: /usr/lib/lua/luci/model/cbi/sqm.lua:18: attempt to index a boolean value stack traceback: /usr/lib/lua/luci/model/cbi/sqm.lua:18: in function 'e' /usr/lib/lua/luci/cbi.lua:48: in function 'load' /usr/lib/lua/luci/dispatcher.lua:696: in function </usr/lib/lua/luci/dispatcher.lua:691>`

Are you using 6rd as well?
As for the webpage errors try using private mode, your browser is probably caching old javascript

vic1234, 2019-09-18:
tried from incognito in chrome same webpage error

what is 6rd?
didt use any specific ipv6 settings

just to let you know my isp enabled ipv6 in the last couple of month

vic1234, 2019-09-18:
a lot of http error in system log
`Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: luci: accepted login on /admin/network/sqm for root from 192.168.1.162
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: Failed to execute cbi dispatcher target for entry '/admin/network/sqm'.
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: The called action terminated with an exception:
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: /usr/lib/lua/luci/model/cbi/sqm.lua:18: attempt to index a boolean value
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/model/cbi/sqm.lua:18: in function 'e'
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/cbi.lua:48: in function 'load'
Wed Sep 18 16:39:17 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/dispatcher.lua:696: in function </usr/lib/lua/luci/dispatcher.lua:691>
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: luci: accepted login on /admin/network/sqm for root from 192.168.1.162
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: Failed to execute cbi dispatcher target for entry '/admin/network/sqm'.
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: The called action terminated with an exception:
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: /usr/lib/lua/luci/model/cbi/sqm.lua:18: attempt to index a boolean value
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/model/cbi/sqm.lua:18: in function 'e'
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/cbi.lua:48: in function 'load'
Wed Sep 18 16:39:38 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/dispatcher.lua:696: in function </usr/lib/lua/luci/dispatcher.lua:691>
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: Failed to execute call dispatcher target for entry '/admin/wireless_assoclist'.
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: The called action terminated with an exception:
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: /usr/lib/lua/luci/tools/status.lua:188: attempt to index a boolean value
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/tools/status.lua:188: in function 'wifi_assoclist'
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/controller/admin/index.lua:233: in function </usr/lib/lua/luci/controller/admin/index.lua:230>
Wed Sep 18 16:45:57 2019 daemon.err uhttpd[1380]: 	(tail call): ?
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: The called action terminated with an exception:
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/index'.
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: A runtime error occurred: [string "/usr/lib/lua/luci/view/admin_status/index.h..."]:20: attempt to index a boolean value
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: 	[string "/usr/lib/lua/luci/view/admin_status/index.h..."]:20: in main chunk
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: 	[C]: in function 'n'
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/template.lua:55: in function </usr/lib/lua/luci/template.lua:47>
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: 	(tail call): ?
Wed Sep 18 16:45:58 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/dispatcher.lua:680: in function </usr/lib/lua/luci/dispatcher.lua:679>
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: Failed to execute call dispatcher target for entry '/admin/wireless_assoclist'.
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: The called action terminated with an exception:
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: /usr/lib/lua/luci/tools/status.lua:188: attempt to index a boolean value
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/tools/status.lua:188: in function 'wifi_assoclist'
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/controller/admin/index.lua:233: in function </usr/lib/lua/luci/controller/admin/index.lua:230>
Wed Sep 18 16:46:08 2019 daemon.err uhttpd[1380]: 	(tail call): ?
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: The called action terminated with an exception:
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/index'.
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: A runtime error occurred: [string "/usr/lib/lua/luci/view/admin_status/index.h..."]:20: attempt to index a boolean value
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: 	[string "/usr/lib/lua/luci/view/admin_status/index.h..."]:20: in main chunk
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:46:09 2019 daemon.err uhttpd[1380]: 	[C]: in function 'n'
[... cut ...]
Wed Sep 18 16:47:35 2019 daemon.err uhttpd[1380]: 	[string "/usr/lib/lua/luci/view/admin_status/index.h..."]:20: in main chunk
Wed Sep 18 16:47:35 2019 daemon.err uhttpd[1380]: stack traceback:
Wed Sep 18 16:47:35 2019 daemon.err uhttpd[1380]: 	[C]: in function 'n'
Wed Sep 18 16:47:35 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/template.lua:55: in function </usr/lib/lua/luci/template.lua:47>
Wed Sep 18 16:47:35 2019 daemon.err uhttpd[1380]: 	(tail call): ?
Wed Sep 18 16:47:35 2019 daemon.err uhttpd[1380]: 	/usr/lib/lua/luci/dispatcher.lua:680: in function </usr/lib/lua/luci/dispatcher.lua:679>
`

Willem1981, 2019-09-25:
> > > SFE is on.
> > 
> > 
> > Yep if you didn't change the mac address it is [00:00:00:00:00:00] meaning it didn't find the correct route it needs to go though.
> > I need to investigate 6in4 further
> 
> Ok, if you need more info or test just let me know.

I have installed the 	Fixed "Add Protocol 41 Handling Tested on 6in4 tunneling" firmware. SFE is working . But it looks like IPV6 is broken.

C:\Windows\system32>ping -6 google.com
Pinging google.com [2a00:1450:400e:80e::200e] with 32 bytes of data:
Request timed out.
Ping statistics for 2a00:1450:400e:80e::200e:
Packets: Sent = 1, Received = 0, Lost = 1 (100% loss),

root@OpenWrt:~# ping ipv6.google.com
PING ipv6.google.com (2a00:1450:400e:80c::200e): 56 data bytes
64 bytes from 2a00:1450:400e:80c::200e: seq=0 ttl=58 time=7.365 ms
64 bytes from 2a00:1450:400e:80c::200e: seq=1 ttl=58 time=7.142 ms
--- ipv6.google.com ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 7.142/7.253/7.365 ms

root@OpenWrt:~# cat /sys/fast_classifier/debug_info | grep '2a00'
o=1, p=6 [ec:d0:9f:d1:41:af]:2a02:0058:00c9:4c00:2dd4:93f3:4c02:f313:45068 2a00:1450:400e:080e:0000:0000:0000:2013:443:[00:30:88:16:ac:fd] m=00000000 h=4

gwlim, 2019-09-25:
> I have installed the Fixed "Add Protocol 41 Handling Tested on 6in4 tunneling" firmware. SFE is working . But it looks like IPV6 is broken.
>
> C:\Windows\system32>ping -6 google.com
> Pinging google.com [...]

is [00:30:88:16:ac:fd] equals to mac address of eth0.2?

Willem1981, 2019-09-26:
No it is not.

gwlim, 2019-09-26:
Can you do ifconfig and show the wan interfaces especially for 6rd?

Willem1981, 2019-09-26:
6rd-wan_6 Link encap:IPv6-in-IPv4
          inet6 addr: ::54f5:94c/96 Scope:Compat
          inet6 addr: 2a02:58:c9:4c00::1/44 Scope:Global
          UP RUNNING NOARP  MTU:1280  Metric:1
          RX packets:30 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19 errors:25 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:16399 (16.0 KiB)  TX bytes:2193 (2.1 KiB)

gwlim, 2019-09-26:
Do you have an interface which have the wan mac address?

Willem1981, 2019-09-26:
eth0.2    Link encap:Ethernet  HWaddr EC:08:6B:E8:1A:E6
          inet addr:xxx.xxx.xxx.xxx  Bcast:84.245.9.255  Mask:255.255.254.0
          inet6 addr: fe80::ee08:6bff:fee8:1ae6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8298 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7319 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1884737 (1.7 MiB)  TX bytes:1612238 (1.5 MiB)

gwlim, 2019-09-29:
Latest firmware doesn't do 6in4/6rd SFE so both your ipv4 and ipv6 should work fine.
You can use fast.com to check for ipv6 speeds and speed test for ipv4 speeds.
I am not sure if ipv6 speeds will be affected.
Pick a fast server.

Willem1981, 2019-09-29:
Ok, But in the earlier build the IPv6 was also working without SFE.
But the speeds are jusst lower. around 300Mbps but my provider gives 1Gbit.

Will SFE and 6RD be working in the new upcomming builds?

[7 more replies]

closed 2019-12-19 by gwlim