gwlim/openwrt-sfe-flowoffload-ath79 #76: UPNP failing

triatomic, 2019-09-03:
On 1043ndv2 with September images upnp fails to add ports.

Tue Sep  3 23:39:23 2019 daemon.err miniupnpd[26485]: Failed to add NAT-PMP 8999 tcp->192.168.1.215:8999 'NAT-PMP 8999 tcp'
Tue Sep  3 23:39:23 2019 daemon.err miniupnpd[26485]: Failed to add NAT-PMP 8999 udp->192.168.1.215:8999 'NAT-PMP 8999 udp'

triatomic, 2019-09-03:
Tue Sep  3 23:37:04 2019 daemon.notice miniupnpd[26485]: HTTP listening on port 5000
Tue Sep  3 23:37:04 2019 daemon.warn miniupnpd[26485]: no HTTP IPv6 address, disabling IPv6
Tue Sep  3 23:37:04 2019 daemon.notice miniupnpd[26485]: Listening for NAT-PMP/PCP traffic on port 5351
Tue Sep  3 23:37:04 2019 daemon.err miniupnpd[26485]: PCPSendUnsolicitedAnnounce() IPv6 sendto(): Bad file descriptor

gwlim, 2019-09-15:
Release reworked test image please test and feedback

closed 2020-05-07 by gwlim