gwlim/openwrt-sfe-flowoffload-ath79 #78: SEPT firmware (Test build)

MadRock88, 2019-09-06:
installed last sept build for 4300
want to install some ipk from tarball
and take this error
 Cannot satisfy the following dependencies for kmod-usb-storage:
 *      kernel (= 4.19.69-1-cc50d72d667dd6eeed2a23087f817934)
 * opkg_install_cmd: Cannot install package kmod-usb-storage.

kernel in software on router  4.19.69-1-67fb2830839ded3e7652d0317c08eb92
on archive with kmodules kernel_4.19.69-1-67fb2830839ded3e7652d0317c08eb92_mips_24kc

do force install or what?)

gwlim, 2019-09-07:
Go back to the version before the Test Image if you don't have any issues with it.

Zsolteeca, 2019-09-07:
I have a problem with the previous version of Faspath when I would like to install qos
I do not know why

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-qos:
 * 	kernel (= 4.19.69-1-cc50d72d667dd6eeed2a23087f817934)
 * opkg_install_cmd: Cannot install package luci-app-qos.

MadRock88, 2019-09-07:
version before - IMHO work slowly, not working port forwarding.
by speed it's version more better, i choose speed )) not ipk)
my IPS give 600mb/s ^)

kolomparrudi, 2019-09-07:
@MadRock88: you'll get this error message if you install kernel modules from openwrt repository. Install them manually from the kmod-package.tar.gz manually

MadRock88, 2019-09-07:
i get this message when i install manually from this package. 
i say that.
in package kernel module have 4.19.69-1-67fb2830839ded3e7652d0317c08eb92 revision
and firmware have same revision, but manual install give warning with revision error on ipk file.
My question was: can i force install this package, maybe difference in kernel not touch this module or not.

Xmister, 2019-09-07:
Your problem is that you try to install a package, which has dependencies, and opkg will try to install that dependency from the official repository.
You need to install everything from the tarball, back to the first package that doesn't have other dependencies.

gwlim, 2019-09-15:
Release reworked test image please test and feedback

MadRock88, 2019-09-16:
On saturday can test. And feedback ofcouse.

MadRock88, 2019-09-21:
new build has new kernel and same error

root@WDR4300:~# opkg install /tmp/usr/lib/opkg/kmod-usb-storage_4.19.72-1_mips_2
4kc.ipk
Installing kmod-usb-storage (4.19.72-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-storage:
 *      kernel (= 4.19.74-1-0518a4bf7833bdf123ece14953088027)
 * opkg_install_cmd: Cannot install package kmod-usb-storage.

in arch tarball kernel version =  kernel_4.19.72-1-54da640ce96c5aae929223a4b7298832_mips_24kc
in software tab = 4.19.72-1-54da640ce96c5aae929223a4b7298832

but same trouble, like previous version.
------------------------------------------------
from pros for new build -  more speed on interface.

gwlim, 2019-12-19:
Are you sure you installed the kmods correctly?
1)Extract the kmods
2)scp the kmods to router /tmp/
3)opkg install /tmp/KMOD_FILE
You need to specify FULL PATH if not it is installing from openwrt upstream repo

MadRock88, 2019-12-20:
yeap, extract, scp to tmp, opkg
error was on sept+ sept test build, dec build not tested 
my main question was:
if I ignore this message, can I install package from tar? 
because i think tar archive was compiled with firmware, and maybe some packages not updated becouse kernel dependencies not changed from previouw kernel for that package.
from other people was same issue on very similar package ( usb/echi)

gwlim, 2019-12-21:
The last time was because there are some issues with using SFE with 6rd services so I recompiled everything unplanned, as a result there might be some kernel mismatch occurring

taz-1, 2020-01-05:
how to install?
from the interface

The installed version of package kernel is not compatible, require 4.19.93-1-4e92f69b… while 4.19.91-1-ad9cf3a6… is installed.

from putty

root@OpenWrt:~# opkg install /tmp/kmod-usb-storage-uas_4.19.91-1_mips_24kc.ipk
Installing kmod-usb-storage-uas (4.19.91-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-storage-uas:
 *      kernel (= 4.19.93-1-4e92f69bcf94fcc1ca5afc513ab5f2b6)
 * opkg_install_cmd: Cannot install package kmod-usb-storage-uas.