gwlim/openwrt-sfe-flowoffload-ath79 #82: SEPT firmware wdr-4300 - (kernel packages archive)

asergiu, 2019-09-19:
Hi gwlim,
   On the sept build the archive with the kernel modules has a mismatch in the hash of of the kernel 
Trying to install kmod-nf-nathelper-extra_4.19.72-1_mips_24kc.ipk from archive:
_opkg install /tmp/kmod-nf-nathelper-extra_4.19.72-1_mips_24kc.ipk
Installing kmod-nf-nathelper-extra (4.19.72-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-nf-nathelper-extra:
 *      kernel (= 4.19.72-1-23e25f21c0f86dda925f0fc261a08d10)
 * opkg_install_cmd: Cannot install package kmod-nf-nathelper-extra._

kernel 
_opkg info kernel
Package: kernel
**Version: 4.19.72-1-54da640ce96c5aae929223a4b7298832**
Depends: libc, libpthread
Status: install user installed
Architecture: mips_24kc
Installed-Time: 1568473905_

Would it be possible to fix the kernel modules archive ? (rebuild module packages with the correct kernel ?)

Thx
Adrian

gwlim, 2019-12-19:
Are you sure you installed the kmods correctly?
1)Extract the kmods
2)scp the kmods to router /tmp/
3)opkg install /tmp/<kmod file>
You need to specify FULL PATH if not it is installing from openwrt upstream repo

asergiu, 2019-12-19:
I extracted them from tar. You answered me at that time ... I will try this evening as it is at my home. Anyway you told me to install another version of your build and on that one it worked.
Adrian

taz-1, 2019-12-21:
I also have the same error when installing packages, I think the error is kernel compatibility

gwlim, 2019-12-22:
The issue should not be present in the December version
Please test before making outdated comments.

closed 2019-12-22 by gwlim

asergiu, 2019-12-22:
Dear gwlim    I am far away from commenting. I do appreciate your work. I only thought my statement would help you pinpoint things in some of the builds. Now I have a version (need to look which one) that works more or less for what I need. If you need me to test a new one (if that would help you) just let me know.Adrian