gwlim/openwrt-sfe-flowoffload-ath79 #85: SFE and DNAT  - AUG build wdr4300

asergiu, 2019-10-05:
Something is broken in the august build of the wdr4300. 
Setting Destination NAT in the GUI to map a wan port say 5000 to an internal IP over TCP does not allow the protocol handshake. TCP handshake seems fine.  
When SFE is disabled all these work - so I assume it has something to do with the way packets are routed through firewall or not (conntrack ?!?!)

Using netcat to check connectivity works (with small packets typed manually) but any other thing (remote desktop, ssh) does not complete the RDP/ssh handshake - so I assume it is after TCP conn is setup. .. Sending some large data DOES NOT work anymore over netcat either !

To test:
1) Enable SFE, setup a DNAT to a machine behind router - on ssh (for example).. Try to connect from the wan side of the router - IT SHOULD NOT work. Connection is reset during ssh handshake !
2) Disable SFE - and keep the same as above and ssh connection now works - you can login and work  !

Same happens for RDP !
Haven't check udp yet

gwlim, 2019-10-06:
Use OCT release instead

asergiu, 2019-10-27:
Cannot load kernel modules in OCT build. For some reason opkg needs 4.19.79 ?!?!? while on the system we have 4.19.75. Odd as the names of the packages imply 4.19.75 - but that could be just a naming error.

opkg install kmod-nf-nathelper-extra_4.19.75-1_mips_24kc.ipk
Installing kmod-nf-nathelper-extra (4.19.75-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-nf-nathelper-extra:
 *      kernel (= 4.19.79-1-b9477a06ef9b0767756d68f06a382e51)
 * opkg_install_cmd: Cannot install package kmod-nf-nathelper-extra.

uname -a
Linux homers 4.19.75 #0 Thu Sep 26 20:15:25 2019 mips GNU/Linux

gwlim, 2019-12-17:
Install kernel modules from the tar.gz

closed 2019-12-17 by gwlim

asergiu, 2019-12-19:
I installed them from .tar.gz
Adrian