gwlim/openwrt-sfe-flowoffload-ath79 #87: Mikrotik RB450G build for next month?

eisengrau, 2019-10-18:
Hi There,

First of all, I would like to thank you for the Archer C7v2 test builds. I had a friend to whom I recommended your builds, he has a v2, and was cursing OpenWrt (and me because I convinced him to flash his device) due to stability issues.  I noticed your recent test build for the C7 v2 exclusively, and he flashed it over straight away, now he's very happy, the stability improved considerably.

My question is that will it be a build for the Mikrotik RB450G like usual, for the next month?

Thank you,
L.

gwlim, 2019-10-19:
Please help by porting all ar71xx to ath79 branch first.
Openwrt upstream has stop supporting all ar71xx targets, Mikrotik belongs to the outdated not supported ar71xx target BUT it can be migrated to the new ath79 targets by updating it to a DTS infrastructure.

eisengrau, 2020-03-15:
Understood. Will research this further.

closed 2020-03-15 by eisengrau