gwlim/openwrt-sfe-flowoffload-ath79 #89: TPLink-1043ND v1 mini build

petakpa1, 2019-10-25:
Hi Gwlim,

Do I understand correctly that if I install your latest OCT2019 minibuild on 1043ND v1 device I will not be able to install additional packages from Luci at all?

Is there any way to integrate the the following services into this minibuild
- Dynamic DNS
- Wake on LAN
- uHTTPd (I use it only fro reaching the Luci from WAN side)
Or inclusion of above are totally impossible because lack of room in the 8 MB flash memory?

gwlim, 2019-12-17:
Not enough space install additional non-kernel packages from openwrt and kernel related packages from the tar.gz

closed 2019-12-17 by gwlim

petakpa1, 2019-12-21:
Hi Gwlim,
Thank you for your answer. This was my assumption too, I understood it.
Is it possible to create a custom build which is based on your DEC2019 SFE-flow offload normal build (which includes Luci), 
but 
- ipv6 support is completely removed (to save space)
+ Dynamic DNS, Wake on LAN and  uHTTPd (I use it only for reaching the Luci from WAN side) is added?

I may donate your work&efforts of course.

I wish you merry Xmas.

Kind regards,
P
From: gwlim 
Sent: Tuesday, December 17, 2019 8:58 AM
To: gwlim/openwrt-sfe-flowoffload 
Cc: petakpa1 ; Author 
Subject: Re: [gwlim/openwrt-sfe-flowoffload] TPLink-1043ND v1 mini build (#89)

Not enough space install additional non-kernel packages from openwrt and kernel related packages from the tar.gz