gwlim/openwrt-sfe-flowoffload-ath79 #98: Where is "Gwlim"? :(

Zsolteeca, 2019-12-11:
Where is "Gwlim"? :(

xchwarze, 2019-12-11:
@Zsolteeca please stop creating junk issues.
gwlim will answer when he has time to do it

Zsolteeca, 2019-12-12:
But has away for about two months :( I think that he hit by the train or whatsever..

vicodin99, 2019-12-12:
@Zsolteeca 

gwlim will answer when he can or update when there's a need. He's a busy person.

Please, stop. You're not helping by spamming junk issues. Your time would be better used if you tried to improve your English. Meanwhile, enjoy these builds as they are, if they're useful to you.

Zsolteeca, 2019-12-12:
You are a clever bullshit vlcodin..

I had to reset the router for a reason.. For example a latest build has really outdated.. Because of software components I use dyndns.. and a few more don't compatible with this firmware because of the environment.. version changes usually two days...  And I can update this whole system in vain because versions will be incompatible with others.. That's why...

closed 2019-12-13 by Zsolteeca