gwlim/openwrt-sfe-flowoffload-ath79 #99: Port forward not working

robinhugy, 2019-12-12:
Port forwards not working for me, portscan says Port Status: closed

WDR4300 | OpenWrt SFE r11009-1cf2495d48

robinhugy, 2019-12-31:
port forward working with dec 2019 image but have sfe offloading problem, its slower than old image
old: stable 350-450 mbit
new: starting with 400+ but a few secs later the max speed is only 250-280 mbit

robinhugy, 2019-12-31:
tried all options in firewall:

SFE flow offloading

Software flow offloading /
Software based offloading for routing/NAT
Hardware flow offloading

0sy, 2019-12-31:
Using pppoe by chance? The overhead is what kills the speed

0sy, 2019-12-31:
> port forward working with dec 2019 image but have sfe offloading problem, its slower than old image
> old: stable 350-450 mbit
> new: starting with 400+ but a few secs later the max speed is only 250-280 mbit

Using pppoe by chance? Some users report slowness when on pppoe, and it's probably due to the overhead.

robinhugy, 2019-12-31:
WAN - Protocol: DHCP client

the config is same like in the old version, nothing changed only firmware

gwlim, 2020-01-02:
port forwarding is not SFE accelerated because there is no NAT

robinhugy, 2020-01-02:
but port forwarding is does not affect speedtest.net

gwlim, 2020-01-05:
Try the latest

robinhugy, 2020-01-05:
oh just see jan 2020, thanks for notice
what "il" means in firmware name? e.g. openwrt-ath79-generic-tplink_tl-wdr4300-v1-**### il**-squashfs-factory.bin

aszorenyi, 2020-01-07:
It means that its for WDR4300 bought in Israel. If you check the product page on OpenWRT site you can see. :)

robinhugy, 2020-01-07:
checked openwrt site but i never seen, thanks Ádám :)

closed 2020-05-07 by gwlim