gwlim/openwrt-sfe-flowoffload-ath79 #100: Feature Request: Allow use of DFS channels in North America

Pentaphon, 2019-12-17:
Any chance we can use DFS channels like we could on the root build?

Here's the old builds that allowed this. https://github.com/infinitnet/lede-ar71xx-optimized-archer-c7-v2

DFS channels are those from 52-144, as shown below. 

[image]

xchwarze, 2019-12-17:
Good to see that they shared the patch folder!
https://github.com/infinitnet/lede-ar71xx-optimized-archer-c7-v2/tree/master/patches
Maybe it would be a good idea to imitate here...

gwlim, 2019-12-17:
Can't you just change your region to a non north america region?

Pentaphon, 2019-12-19:
I didn't realize that was possible on the US-specific firmware for the Archer C7v2.

gwlim, 2019-12-20:
I think hostapd respects region setting
Also the github which you linked does not contain any of the said patches.
In fact those patches were derived from my old repositories.

Pentaphon, 2019-12-20:
Changing to a non-US region just disables my wireless interface on my Archer C7v2. I had to make it driver default or US for it to work again.

gwlim, 2019-12-21:
When I do the command below there should be nothing blocking DFS
```
root@openwrt:~# iw reg get | tail -n 18
phy#1
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

```

gwlim, 2019-12-21:
Anyway the builds are compiled with
`CONFIG_ATH_USER_REGD=y`

Pentaphon, 2019-12-21:
> When I do the command below there should be nothing blocking DFS

When I switch to say, channel 52, on Qualcomm Atheros QCA9880 802.11nac, the network is no longer accessible and does not even show up in my list of available networks on both my phone and laptop. 

This is on the latest DEC build for the Archer C7v2 US. I could access channel 52 with the infinitnet builds, but not yours. 
[image]

gwlim, 2019-12-21:
That is because nothing has associated yet.
Wait for the wifi to start then associate a device

Pentaphon, 2019-12-21:
It's been nearly 30 minutes and my AC wifi just stays like it does in the screenshot I posted. I can't see it on my phone, laptop, anything so I can't associate it. It springs back to life when I go to say, channel 36.  Even a full reboot when on channel 52 does not make it work. Only non-DFS channels work on my Archer C7v2. What device are you using?

gwlim, 2019-12-21:
I am using Archer C7v4 (US) on the box.
Maybe you can start backing up your art partition and start trying art partition from other devices

Pentaphon, 2019-12-21:
>Could be your client devices don't recognise DFS channels?

No, because they all worked fine when I used DFS channels on the infininet builds. Something about those builds made every DFS channel work for me. Last build I used where DFS worked perfectly was the latest build from this fork. 
https://github.com/shunjou/openwrt-optimized-archer-c7-v2

gwlim, 2019-12-21:
Anyways I am using DFS Channel 100 on my Wireless Mesh Links so yes it works for me

Pentaphon, 2019-12-21:
>Try resetting wifi config

No luck. The only thing that allows me to use DFS on my ArcherC7v2 is flashing this fork. 
https://github.com/shunjou/openwrt-optimized-archer-c7-v2

gwlim, 2019-12-21:
My experience is that openwrt wireless config is flaky, for dual band devices wireless device sequence matter so always try to wifi config and reset to default config and then reconfigure from there

Pentaphon, 2019-12-21:
> DFS takes around 2 minutes to reinitialize

Actually DFS was the same as non-DFS channels on that fork I linked before. All channels from 36 to 165 started working at the same time when experimenting. Is there any way to "unlock" DFS on your builds the way that fork did? I can't explain why it worked so well but it did.

gwlim, 2019-12-21:
I am pretty sure there is nothing he did which I didn't do.
Most, if not all the patches he had was from my repo.
I hope you did try to rescan your wifi client devices to verifiy it is not broadcasting in DFS mode
It used to be regdb.txt had to be modified.
But it is no longer needed in newer openwrt if you set
`CONFIG_ATH_USER_REGD=y`

Pentaphon, 2019-12-22:
> But it is no longer needed in newer openwrt if you set
>CONFIG_ATH_USER_REGD=y

Sorry but how do I set this?

gwlim, 2019-12-22:
> 
> 
> > But it is no longer needed in newer openwrt if you set
> > CONFIG_ATH_USER_REGD=y
> 
> Sorry but how do I set this?

It is already enabled which is why DFS works.

Pentaphon, 2019-12-22:
> It is already enabled which is why DFS works.

I'm not trying to be a nuisance, but it really doesn't work on your ArcherC7v2 build, possibly because it is US specific. I've tried to get it to work several times this week. Hopefully somebody else with an ArcherC7v2 in the US will confirm my observations.

gwlim, 2019-12-22:
Another alternative is changing other versions the ath10k-ct firmware (988x) from
https://www.candelatech.com/ath10k-10.1.php
How to use is from the page itself.

Pentaphon, 2019-12-22:
> Another alternative is changing other versions of the ath10k-ct firmware (988x) from
> https://www.candelatech.com/ath10k-10.1.php
> How to use is from the page itself.

No such luck. I tried an older version, then ct, then the original ath10k firmware, making sure I only had 1 firmware installed. All shut down when I switched to channel 52.  I tried those 3 options as soon as I read your post. 

Meanwhile, in the last 10 minutes, I decided to flash the https://github.com/shunjou/openwrt-optimized-archer-c7-v2 build into my ArcherC7v2 to see if it would work. It did. Here's a screenshot of me running that shunjou firmware on channel 52. I cannot explain why it works perfectly on that firmware but not yours. I even made sure to upgrade to the latest 2019-10-03 ath10k firmware via the software page and it still works fine. 

[image]

gwlim, 2020-01-11:
Upstream bug reference 
https://bugs.openwrt.org/index.php?do=details&task_id=1920&string=firmware

Pentaphon, 2020-01-22:
> Upstream bug reference
> Initialization sometimes will get stucked, just have to try various channels until 1 that works
> https://bugs.openwrt.org/index.php?do=details&task_id=1920&string=firmware

Thanks for looking into this. I'm not sure this bug has to do with my issue because 5ghz does not work no matter how hard I try when using a DFS channel. Rebooting, etc.

MrObvious, 2020-04-17:
According to https://bugzilla.redhat.com/show_bug.cgi?id=1671935 this is fixed in more recent ath10k firmware?

gwlim, 2020-04-17:
It might not be firmware issue, if possible pass a copy of your art partition

MrObvious, 2020-04-17:
Do you want it while I’m set to a DFS channel?

gwlim, 2020-04-17:
art partition does not change, just trying to check if the art partition of V2 has any difference with V4

MrObvious, 2020-04-17:
Attached. 

[art.zip](https://github.com/gwlim/openwrt-sfe-flowoffload/files/4493739/art.zip)

gwlim, 2020-04-18:
[archer_c7v4_art.bin.zip](https://github.com/gwlim/openwrt-sfe-flowoffload/files/4496346/archer_c7v4_art.bin.zip)
It is different from Archer C7v4 if you are brave enough you can try it

MrObvious, 2020-04-19:
Not sure. I’ll try overclocking probably first since there have been users who have successfully done it.

MrObvious, 2020-06-28:
Update, updated to May 2020, didn’t keep existing settings, and channel 100 works now. 
[image]

closed 2020-06-29 by gwlim