gwlim/openwrt-sfe-flowoffload-ath79 #101: Unable to install software packages :(

Zsolteeca, 2019-12-18:
I tried to install the dyndns package to the new firmware (dec2019) and I got the following error

"RPCError

RPC call to luci.ddns/get_services_status failed with error -32002: Access denied
  at handleCallReply (http://192.168.1.88:13360/luci-static/resources/rpc.js?v=git-19.348.21714-7b0ae11:12:3)"

:(

gwlim, 2019-12-18:
Upstream Luci based apps are broken due to Luci transitioning from Lua backend to JS backend.
It is not likely to get fix anytime soon.
Workaround: use commandline to configure dyndns

Zsolteeca, 2019-12-18:
Thank you

I found a solution to this problem if anyone would face

"Try

chmod +x /usr/libexec/rpcd/luci.ddns
service rpcd restart"

And now the ddns service is working :)

Zsolteeca, 2019-12-18:
I have found this solution which partially fix this software component..

"Please try

chmod +x /usr/libexec/rpcd/luci.ddns
service rpcd restart"

closed 2019-12-18 by Zsolteeca