gwlim/openwrt-sfe-flowoffload-ath79 #104: JAN 2020 sysupgrade image for 1043ND v2 issue

0sy, 2019-12-29:
https://github.com/gwlim/openwrt-sfe-flowoffload/blob/master/JAN-2020/openwrt-sfe-flowoffload-normal/TP-Link%20WR1043NDv2-mips74k-ath9k/openwrt-ath79-generic-tplink_tl-wr1043nd-v2-squashfs-sysupgrade.bin

Ends up with https://i.imgur.com/L0Wuzzo.png

elgab, 2019-12-29:
The files could get corrupted, maybe @gwlim can help us in the future with some SHA256 or GPG signatures.

yonizaf, 2019-12-30:
I tried the specified file. It flashed and worked fine.

Zsolteeca, 2019-12-30:
I also flashed the firmware I have the same modell of the router what you have.

0sy, 2019-12-31:
Yeah, I tried again and it worked. Weird how the first two attempts failed.

OPENWRT_RELEASE="OpenWrt SFE r11797-1579599079"

closed 2019-12-31 by 0sy