gwlim/openwrt-sfe-flowoffload-ath79 #105: WDR4300 v1 can't associate more than one radio network since 2020 builds

dexterxx-pl, 2020-01-08:
Since 2020 builds (JAN-2020 and FEB-2020) radio in TP Link WDR4300 cannot associate more than one network on same radio.

**Problem:**

Now:
[image]

"Wireless is not associated"
In interfaces section got "error: network device is not present".

Disabling first network on list magically fixes that second...
[image]

**Before, OCT-2019 build**
[image]

Tried:
* forcing BSSID/MAC on interfaces/networks -- also since 2020's builds openwrt detect's other base-MAC on my device (last MAC's position -2)
* changing settings between networks to more similar
* _forgot..._

Probably is base openwrt changes (lot of changes since OCT-2019 build: at least new kernel...).

Any advice?

gwlim, 2020-01-09:
Check how to reset /etc/config/wireless and reconfigure using LuCI

Zsolteeca, 2020-01-09:
Same problem to me on my Tplink WR1043 ND v2.1 device..
I think that it is an Openwrt based problem  :(
The wireless network named xy is working but I can't see the device(s) on the network from Luci.

dexterxx-pl, 2020-01-10:
Regenerating configuration doesn't help. I've also updated first post mentioning, that 5GHz Radio associates much longer time than i previous builds... but that's not so big problem.

Btw. configuration comparison, there's no crucial differences:

**Mine vs BASE** 
[image]

**Mine vs ~almost LuCI reconstruction**
[image]

dexterxx-pl, 2020-01-10:
I've flashed nowadays new stock 19.07.0 OpenWRT -- [openwrt-19.07.0-ath79-generic-tplink_tl-wdr4300-v1-squashfs-sysupgrade.bin](https://downloads.openwrt.org/releases/19.07.0/targets/ath79/generic/openwrt-19.07.0-ath79-generic-tplink_tl-wdr4300-v1-squashfs-sysupgrade.bin)

Radio (at least 2,4GHz which I tested) **works with many network**.
```
Model	TP-Link TL-WDR4300 v1
Architecture	Atheros AR9344 rev 2
Firmware Version	OpenWrt 19.07.0 r10860-a3ffeb413b / LuCI openwrt-19.07 branch git-20.006.26738-35aa527
Kernel Version	4.14.162
```
Config for comparison - in this case **I had to** regenerate, because LuCI showed .... three radios.
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel 'auto'
        option country 'PL'
        option hwmode '11g'
        option path 'platform/ahb/ahb:apb/18100000.wmac'
        option htmode 'HT20'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'XXXXXX'
        option encryption 'psk2'
        option key 'XXXXXX'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel 'auto'
        option country 'PL'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0'
        option htmode 'HT20'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'XXXXXX'
        option encryption 'psk2'
        option key 'XXXXXX'

config wifi-iface 'wifinet2'
        option network 'guest'
        option ssid 'XXXXXX'
        option encryption 'psk2'
        option device 'radio0'
        option mode 'ap'
        option isolate '1'
        option key 'XXXXXX'
```

I'm gonna check tomorrow does changing path fixes something in @gwlim builds.
Maybe build from stable, than trunk, would be better idea?

gwlim, 2020-01-11:
Your path on the 2.4G is wrong.
It should 
 option path 'platform/ahb/18100000.wmac'
Not sure how you received that value in the first place
Did you reset the wifi config correctly?

Zsolteeca, 2020-01-11:
The private mode didn't work for me. The result is the same to DexterX..

gwlim, 2020-01-11:
Anyway my config looks something like this
```
root@openwrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'platform/ahb/18100000.wmac'
        option country 'US'
        option legacy_rates '0'
        option noscan '1'
        option channel '3'
        option htmode 'HT40'
        option txpower '21'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'SSID1'
        option skip_inactivity_poll '1'
        option disassoc_low_ack '0'
        option key '*******'
        option ieee80211w '1'
        option encryption 'psk2+ccmp'
        option network 'lan'
        option ft_psk_generate_local '1'
        option mobility_domain '1234'
        option ieee80211r '1'
        option ft_over_ds '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0'
        option country 'US'
        option legacy_rates '0'
        option htmode 'HT40'
        option noscan '1'
        option channel '64'
        option txpower '20'

config wifi-iface 'mesh0'
        option device 'radio1'
        option key '*******'
        option mode 'mesh'
        option mesh_rssi_threshold '0'
        option disassoc_low_ack '0'
        option mesh_id 'MESH01'
        option encryption 'sae'
        option skip_inactivity_poll '1'
        option mesh_fwding '0'
        option network 'lan'
        option noscan '1'

config wifi-iface 'wifinet2'
        option network 'lan'
        option skip_inactivity_poll '1'
        option ssid 'OpenWrt'
        option encryption 'psk-mixed+ccmp'
        option device 'radio0'
        option mode 'ap'
        option key '*******'

```

gwlim, 2020-01-11:
btw the way to reset wifi config is
rm /etc/config/wireless
reboot
wifi config

Axius, 2020-01-11:
What kernel is being used in the FEB-2020 build?

gwlim, 2020-01-11:
> What kernel is being used in the FEB-2020 build?

Should be 4.19.91

closed 2020-05-07 by gwlim