gwlim/openwrt-sfe-flowoffload-ath79 #107: Archer C7 v2 Clients not Showing up in Luci

Go0oSer, 2020-01-12:
Beginning with the Dec 2019 release, I have noticed a number of connected clients that wont show in Luci on either the Status Page or Network > Wireless page.  Same issue persists across 2 different Archer C7 v2 pieces of hardware.

gwlim, 2020-01-19:
If you are using it in a mesh or multiple AP joined by wired backend configuration, you must 
1)Disable DHCP on all AP except the main router
2)Associated Clients will only appear on the AP it has associated locally but will register itself in the main DHCP of the core router

Go0oSer, 2020-01-19:
Neither Archer C7v2 is responsible for DHCP.  I use PFSENSE as my upstream router.  These units are configured as Access Points.  ODHCP is disabled from startup and is stopped on both of them.  19.07 release does not exhibit the same issues.

closed 2024-08-28 by Go0oSer