gwlim/openwrt-sfe-flowoffload-ath79 #118: Luci menus on login

Zsolteeca, 2020-03-27:
I have flashed the latest version of this firmware it works well but when I would like to login to luci the menus appearing on the top and I can navigate within the menus but nothing will do (luckily :D )
It is a bit strange to me I probed with other browsers.. Chrome,Mozilla,Opera.. and Luci has got the same "appearing".. 

I think about this:

Luci look like this [https://ibb.co/RQ9xnpG](url)

Thank you for your firmware!

gwlim, 2020-03-28:
Yea it is a upstream thing, I didn't change anything on LuCI side.
Can't really call it a bug because to access the pages you still need to authenticate.

metalrise, 2020-03-29:
I also flashed the 2020 MAR build on TL-WR1043ND v2, and the Wifi menu is missing.

I think that they fixed some things and broken another

Zsolteeca, 2020-03-29:
> 
> 
> I also flashed the 2020 MAR build on TL-WR1043ND v2, and the Wifi menu is missing.
> 
> I think that they fixed some things and broken another

So did you say that the Network/Wireless menu isn't accessible?

metalrise, 2020-03-30:
Yes, but it's not an software issue with openwrt, its a hardware issue with the router.

Tomorrow I have to connect the serial console and see what is the problem

gwlim, 2020-03-30:
Check dmesg and logread.
Make sure wifi initializes
And check your art (atheros radio test) partition
If you need to reset /etc/config/wifi do it.

metalrise, 2020-04-04:
"He's dead, Jim!"

Tried tftpd recovery with openwrt-factory and with stripped stock firmware, no luck.

The radio died after pressing the reboot button in Luci. And the fun part part is that the Winbond 25Q64FV memory can be programmed in CH341A only with hardware modifications to the programmer.

One last attempt is to write another dump to the memory, but I'm 90% sure that the radio is dead

gwlim, 2020-04-05:
A

> "He's dead, Jim!"
>
> Tried tftpd recovery with openwrt-factory and with stripped stock firmware, no luck.
>
> The radio died after pressing the reboot button in Luci. And the fun part part is that the [...]

On Openwrt flast firmware Menu download the art partition and open it with a hex editor.
Make sure that the art partition has valid data and present.
art partition cannot be restored by flashing to stock, you need to get it from another identical unit if it is overwritten and lost

gwlim, 2020-05-07:
Fixed, however luci-app-wireguard is not working because it needs to be updated and the maintainer has not done so

closed 2020-05-07 by gwlim