gwlim/openwrt-sfe-flowoffload-ath79 #120: SQM: ERROR: QDISC cake is NOT useable (after update)

staimalex, 2020-04-08:
Hi, i have a WR1043N v5 using OpenWrt SFE r1270
I came up with the stupid idea to update to the march build and now sqm doesn't work anymore:
i see this in the log, but i don't really understand what his problem is:

Wed Apr 8 12:59:08 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
Wed Apr 8 12:59:08 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: Invalid argument
Wed Apr 8 12:59:08 2020 user.notice SQM: ERROR: QDISC cake is NOT useable.
Wed Apr 8 12:59:08 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
Wed Apr 8 12:59:08 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: Invalid argument
Wed Apr 8 12:59:08 2020 user.notice SQM: ERROR: QDISC cake is NOT useable.

has somebody an idea what is happening?

gwlim, 2020-04-09:
This happened
https://git.openwrt.org/?p=openwrt%2Fopenwrt.git&a=search&h=HEAD&st=commit&s=cake

gwlim, 2020-05-07:
Fixed

closed 2020-05-07 by gwlim