gwlim/openwrt-sfe-flowoffload-ath79 #121: Archer C7 v2 Best Wi-Fi Speed?

MrObvious, 2020-04-15:
Hi. Just wondering if I've enabled everything. I'm kind of new to OpenWRT so I am trying this build and it seems to work well! I've gotten 900 Mb/s down from Ookla with my Cox Gigablast, but I'm wondering if Wi-Fi will max out around 350 Mb/s? I'm testing on a Thinkpad P52 and iPhone 8. It seems CPU maxes out using top on ssh, but I'm not sure if there is more I can do?

I'm using 5 GHz, SFE offloading (seems better than software flow with hardware flow offloading), channel 157, AES WPA2. I'm open to suggestions, or if this is all this device is capable of on this build that's fine.

gwlim, 2020-04-16:
you shouldn't be testing while SSH in at the same time while testing on a single core cpu router.
I have tested that on wireless ac it does goes up to maximum of 500Mbps at close range single client connected. If you are using Intel wifi don't forget to update your wireless driver

https://www.intel.com/content/www/us/en/support/articles/000017246/network-and-i-o/wireless-networking.html

MrObvious, 2020-04-16:
Thank you for the response. I wonder what settings I'm missing? I really haven't changed more than the wifi settings on this router and the SFE. I tried to disable all the ipv6 boxes I could find.

MrObvious, 2020-04-16:
`root@OpenWrt:/etc/config# cat wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0'
        option htmode 'VHT80'
        option legacy_rates '0'
        option channel '157'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '******'
        option skip_inactivity_poll '1'
        option disassoc_low_ack '0'
        option key '******'
        option encryption 'psk2'
        option ieee80211w '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'platform/ahb/18100000.wmac'
        option htmode 'HT40'
        option legacy_rates '0'
        option beacon_int '200'
        option channel '1'
        option disabled '1'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'ap'
        option encryption 'psk2+ccmp'
        option skip_inactivity_poll '1'
        option disassoc_low_ack '0'
        option key '******'
        option network 'lan'
        option ssid 'TP-LINK_1249_5G'
        option disabled '1'
root@OpenWrt:/etc/config# cat dhcp

config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        list dhcp_option '6,1.1.1.1,8.8.8.8'
[... 155 lines cut ...]

config interface 'wan'
        option ifname 'eth0.2'
        option proto 'dhcp'
        option ipv6 'off'
        option peerdns '0'
        list dns '1.1.1.1'
        list dns '8.8.8.8'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '2 3 4 5 0t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '1 6t'`

That should have most of my config items. I am not using SQM. I am not overclocked either.

gwlim, 2020-04-17:
option noscan '1'
And try switching to DFS Frequencies.

MrObvious, 2020-04-17:
DFS frequencies don't load for me. I'll try overclocking soon.

gwlim, 2020-04-17:
Problem is that DFS is working on my Unit, I cannot fix what I cannot see

[image]

closed 2020-05-07 by gwlim